VHDL-LS/rust_hdl #6: Visibility rules

kraigher, 2018-11-24:
The next step is to work on visibility rules for use clauses and nested scopes.
I have identified the following rules.

### Immediate scope always takes precedence for non-overloaded
```vhdl
constant ident : natural := 0;
use work.pkg.ident;
-- local ident trumps use
constant ident2 : natural := ident;
```

### Multiple non overloaded names require disambiguation 
```vhdl
use work.pkg1.ident;
use work.pkg2.ident;
-- Error requires disambiguation event if pkg1.ident and pkg2.ident are different types
-- Also true if .all was used
constant ident2 : natural := ident;

-- The following would work without ambiguity though
use work.pkg1.ident;
constant ident2 : natural := ident;
use work.pkg2.ident;
```

benreynwar, 2018-12-04:
I'm trying to get my head around how a semantic pass might go.

* Step through the AST of a design unit.
* A DeclarativeRegion structure keeps track of what symbols have been declared in this region so far.
* As we process declarations we add the new symbols to the Declarative Region.
* As we process symbol uses we check if the symbols have already been declared in this region.  If they haven't we check in parent regions recursively.  If the declaration is present in this design unit we link the use of the symbol to it's declaration.  If the declaration is not present we add it to a set of unresolved symbols along with the set of packages that were in scope at the time of use.

* Step through all design units

* Go back to each design unit and resolve the unresolved symbols.  Identify circular dependencies.  Link symbol uses with their declarations in appropriate packages.  Identify ambiguous symbol uses.

Is this the  general direction you were thinking?
At the moment it looks like the semantic analysis is making sure declarations are consistent with one another but hasn't worried about symbol uses yet.  I that correct?

kraigher, 2018-12-04:
My current thinking is:

- Traverse the entire AST and build DeclarativeRegions. The DeclarativeRegions only live temporarily in this pass and are thrown away afterwards. Public declarative regions such as in packages which might be saved for later use.
- Desing units are only analyzed once and will be analyzed on-demand (See 2. in #7) 
- Check for duplicate declarations as you go. 
  - **DONE**
- Check some rules regarding protected type declaration/body deferred constant and incomplete type when a region is complete.
  - **DONE**
- Check for missing declarations in all names and use clauses. 
   - **Partially done for use clauses, this is the next step**

The there are two options to proceed:
1. Annotate back to the AST (by mutating it) the multiple potential references for each name and perform type checking in a separate pass.
2. Perform type checking in the same pass and write back to the AST the single resolved reference.

Also note that I use a Test-Driven-Development workflow which means I have an idea of where I want to go but I do not go ahead and build everything I think I will need without having tests that require it. Thus the current implementation may have shortcomings compared to my final intention since I have not defined the test cases that require the final intention. This is also the reason I have been holding off adding new fields to the AST since I want tests to require it and I do not want to make premature changes to the parsed AST (which is well understood by me) with semantic information (where I am stilll building my mental model of how it will work).

My strategy is also to develop the semantic analysis such that the example project always continues to work without errors. This means I will try not to implement partial solutions that will cause false errors but rather partial solutions that produce false positives. In this way the language server will never be unusable and provide incrementally more value until it is "complete".

benreynwar, 2018-12-04:
OK.  So when we hit a "use work.mypkg.all", we pause the analysis of the design unit, and go and parse the appropriate package.   Similarly if we hit a "myblah: entity work.blah" we pause the analysis, and go and parse that entity.  This allows the semantic analysis to be done in a single pass, but reduces the possibility for parallelism.  Is this implemented already?  I didn't see it there.

kraigher, 2018-12-04:
The on-demand analysis is implemented in my local repository. I will most likely push it today. This method will still allow enough parallelism. You can always start analysis for many design units in parallel and use locking to avoid duplicate analysis or race conditions. Only sequential dependencies would not get analysed in parallel due to locking but for large projects where parallel speed matters the dependency graph has high fan-out so there should be plenty of non-serial dependencies.

kraigher, 2018-12-05:
I pushed the first version of the on-demand analysis now.

closed 2022-12-07 by kraigher