VHDL-LS/rust_hdl #7: Analysis order

kraigher, 2018-11-26:
A goal of the analyser is to automatically compute the analysis order of design units such that the user does not have to care about it.

Automatically computing the analysis order requires traversing the entire AST and finding all use clauses, packages instances as well as instantiations. Circular dependencies might occur which should trigger an error.

### Simple case
```vhdl
use lib.unit;
```
Simple case, introduces depenency on `lib.unit`

### Harder case
```vhdl
use lib.all;
```
Trickier case especially if `lib` is `work`. All names within `lib` cannot be considered a dependency automatically since it might create circular dependencies especially for `work` but also with co-dependent libraries.
We will have to check which names are actually used and only then create the dependency.
```vhdl
use lib.all; -- contains pkg
use pkg.foo; -- Only here we create the depenency on lib.pkg
```
#### Corner case 
This also unfortunately interacts with normal declarative part analysis, for example.
```vhdl
use lib.all; -- contains pkg
package pkg is new gpkg;
use pkg.foo; -- foo might be a nested package instance within pkg
-- Should we consider lib.pkg a dependency here, what if pkg was a local constant?
```
Maybe it is ok to falsely introduce a dependency here to simplify the logic without any practical problem. 

## Potential solutions
I see two solutions to making automatic analysis order work.

### 1. Separate pass to compute the analysis order
Have a separate pass to explicitly only compute the analysis order.

*pros*
- Clean with an explicit pass to extract dependencies. 
- Can be used directly to emit a compile order for use by other tools
- Can be used to determine when things do not need to be re-analysed in case of incremental changes.
- Might be used to compute opportunities for parallel analysis.
*cons*
- A separate pass costs performance
- Potentially duplicate code with normal declarative part analysis

### 2. Compute analysis order with recursion
Compute the analysis order concurrently with other analysis implicitly. If the use of an unanalysed unit is encountered we re-curse and analyse the unit. Units will have to be flagged to detect circular dependencies.

*pros* 
- Single pass
- Can share code with normal scope analysis

*cons*
- Maybe will run into stack overflow for large project if unlucky
- Analysis order is not explicitly computed and extra structure must be created to extract it
- Analysis order can not be had without running full analysis
- Analysis order is not known before starting full analysis which might inhibit parallelism

kraigher, 2018-11-26:
Users of explicit compile order are third party tools like GHDL or commericial simulators that want to have the files in compile order.

Option 3. is probably a better variant of option 2. but with an explicit stack/queue instead of using function call stack. I think it is worth to consider for the future. 

Regarding option 1. I think it can be made very simple if you accept extra dependencies in the corner case above which I think is not a big practical problem. Thus I think option 1. is the simplest one to implement for the first version. Having the analysis order before analysis will make it less complex to implement than doing something like 3. Doing 3 could become an option in the future as either a performance optimization or if the corner case above becomes practically interesting.

I want to focus on getting analysis right before I try to optimize performance to much. It the old rule of first making it correct then making it fast.

FenrisTC, 2018-11-26:
> Users of explicit compile order are third party tools like GHDL or commericial simulators that want to have the files in compile order.

Right. I'm not arguing against the functionality of generating compile order outputs. I just think that it's not much of an issue to compute the compile order explicitely for this purpose, if it isn't needed in for the analysis itself (i.e. Option 2 & 3).

> Option 3. is probably a better variant of option 2. but with an explicit stack/queue instead of using function call stack. I think it is worth to consider for the future.
>
> Regarding option 1. I [...]

I might be very wrong here, but I'm not so sure about option 1 being simple (even with the relaxed corner case dependency). It seems to me, that option 1 is by default harder to implement than option 2, since 1 models dependencies explicitly, while 2 (and 3) only care about the existence of the symbols they import.
For 2 you'd just call the analysis procedure on the design units that provide your `use` clauses.

> 
> I want to focus on getting analysis right before I try to optimize performance to much. It the old rule of first making it correct then making it fast.

I agree with this.
Option 3 is probably too much effort for an implementation to get things running.
But I also think that a later, more optimised implementation would probably go similar route as option 3, i.e. single pass without an explicit dependency representation.
This makes the architecture fairly different from option 1, making a conversion later on harder ,especially if user facing features like compile order files depend on this architecture.

kraigher, 2018-11-26:
The hard thing with 3. is that you could be forced to swap out at any nested position where you could be in the middle of several declarative regions. For example in nested function definitions. You would have to keep track of the continue-position for each parent region so the continue-position cannot be a single value. With option 2. this is done implicitly with the loop counter for each region on the stack.

I have started going down the option 1 path now but I am still thinking about if option 2 is better. What I like about 1 is the separation of concerns in that I can isolate one part that only cares about dependencies between design units and another part that can analyse design units relying on that the order is already correct. Maybe 2 is better because I will not have to re-visit all names and recompute the visibility. I think I will try 2 and see how it feels.

kraigher, 2018-12-06:
I have implemented option 2. now with on-demand analysis and circular dependency detection. It works well.

closed 2018-12-06 by kraigher