VHDL-LS/rust_hdl #10: Create NeoVim integration

kraigher, 2018-11-28:
Create NeoVim integration of VHDL language server and publish instructions on how to use it.

abyszuk, 2019-01-19:
Hi,
I've managed to get the server working with Vim 8.1 and 'vim-lsp' plugin.
You have to add following code to vimrc:
```vim
" VHDL
if executable('vhdl_ls')
    au User lsp_setup call lsp#register_server({
        \ 'name': 'vhdl_ls',
        \ 'cmd': {server_info->['vhdl_ls']},
        \ 'root_uri':{server_info->lsp#utils#path_to_uri(lsp#utils#find_nearest_parent_file_directory(lsp#utils#get_buffer_path(), 'vhdl_ls.toml'))},
        \ 'whitelist': ['vhdl'],
        \ })
endif

" vim-lsp signs are very useful in spotting bugs
let g:lsp_signs_enabled = 1
let g:lsp_diagnostics_echo_cursor = 1
```

Syntax checking triggers only after saving file, it's not enough to just leave insert mode.

suoto, 2019-10-17:
The LSP side seems to be a bit off except when using with VS Code. Tested with https://github.com/dense-analysis/ale and https://github.com/autozimu/LanguageClient-neovim and diagnostics will usually show up upon opening a file, but at some point they seem to freeze. Adding intentional errors or commenting out a line with error seemingly don't make any difference.

Couldn't not get the completion to work.

All tests used `vhdl_ls` built locally from 3d9f1c8.

For the record, set ALE up with

```viml
call ale#linter#Define('vhdl', {
\   'name': 'rust-hdl',
\   'lsp': 'stdio',
\   'language': 'vhdl',
\   'executable': 'vhdl_ls',
\   'command': 'vhdl_ls',
\   'project_root': getcwd(),
\ })

let g:ale_linters['vhdl'] = ['rust-hdl']
```

And LanguageClient-neovim with

```vim
let g:LanguageClient_serverCommands = {
\ 'vhdl': ['vhdl_ls', ],
\ }
```

kraigher, 2019-10-17:
I can have a look. The LSP protocol involves the client and server to declare capabilities. Maybe vhdl ls or vim ignores some important capability setting. For example vhdl ls declares that it wants the entire file sent for every change to avoid implementing the file diff management at this point. If that is ignored by vim it will only work when first opening the file since then it will send everything always. It should be possible to see by logging the jsonrpc traffic.

suoto, 2019-10-17:
https://github.com/autozimu/Language Client-neovim allows dumping the json traffic (and events) to a log, I'll upload them tomorrow. Does rust hdl logs its activities zokewhe) somewhere?

suoto, 2019-10-18:
Added a trace using https://github.com/neoclide/coc.nvim
[trace.txt](https://github.com/kraigher/rust_hdl/files/3743952/trace.txt)

It seems that rust hdl is not responding capabilities properly:
```
[Trace - 13:57:51] Received response 'initialize - (0)' in 17ms.
Result: {
    "capabilities": {
        "textDocumentSync": 1
    }
}
```
and therefore the client does not send any completion request to the server.

kraigher, 2019-10-18:
Well vhdl_ls does not support completion yet so it does not list that as a capability. From your trace it seems to be working. The client sends textDocument/didChange with full content for each change and the server responds with textDocument/publishDiagnostics

suoto, 2019-10-18:
OK, I messed up things, it was another extension I didn't realized was active providing completion (I'm not a regular VS Code user), sorry about that.

I'll try to check the LSP messages in and out to try and isolate is the other issue (diagnostics not always showing) is a problem in the client I'm using or on rust hdl.

Ahmad-Zaklouta, 2020-02-25:
Hi!
Is it possible to list instructions on how to use it with Neovim. I am new to Neovim and linux and I have been trying to have a good setup for VHDL development. I have been trying with hdl_checker and other tools but could not manage to run any of them.

befedo, 2020-02-25:
> Hi!
> Sorry if this is not relevant to this issue.
> Is it possible to list instructions on how to use it with Neovim. I am new to Neovim and linux and I have been trying to have a good setup for VHDL development. I have been trying with hdl_checker and other tools but could not manage to run any of them.

Hi there,

At first you need lsp integration. There are many plugins available, like [nvim-lsp](https://github.com/neovim/nvim-lsp) or [coc.nvim](https://github.com/neoclide/coc.nvim). Then you could configure rust-hdl like any other lsp server, have a look at the examples for your chosen plugin.

pidgeon777, 2021-01-27:
This is actually what I'm using in my Neovim coc-settings.json file for usage with coc.nvim:

````json
"languageserver": {
	"rustHdl": {
		"command": "vhdl_ls",
		"args": [""],
		"filetypes": [
			"vhdl"
		]
	}
}
````

However, when coc.nvim starts it is stuck into this state:

[image]

How to solve this?

If possible it would be nice if someone could share the configuration for using rust_hdl with Vim/Neovim.

SethGower, 2021-07-14:
Late to the game, but figured I'd share this. With neovim's built-in LSP, using `neovim/nvim-lspconfig` this is the custom server declaration I made, and it works. 
```lua
if not lspconfig.rust_hdl then
  configs.rust_hdl = {
    default_config = {
      cmd = {"vhdl_ls"};
      filetypes = { "vhdl" };
      root_dir = function(fname)
        return util.root_pattern('vhdl_ls.toml')(fname)
      end;
      settings = {};
    };
  }
end
```

pidgeon777, 2021-07-19:
> Late to the game, but figured I'd share this. With neovim's built-in LSP, using `neovim/nvim-lspconfig` this is the custom server declaration I made, and it works.
>
> ```lua
> if not lspconfig.rust_hdl [...]

I confirm this works.

Xcodo, 2021-07-19:
@SethGower @pidgeon777 is there a way to create an extension or similar to package this solution up in NeoVim? If not I'd suggest we add this solution to the getting started.

SethGower, 2021-07-19:
@Xcodo to be honest, I would recommend doing both. We (I say "we", since I am planning on starting to contribute more) add it to the getting started, but I also planned on making a PR over at [lspconfig](https://github.com/neovim/nvim-lspconfig) to add this. Only reason I haven't yet, is `rust_hdl` is not reliable to install (last I checked). Since the `cargo install rust_hdl` fails. I have a temp fix on my system where I just cloned it, built it with `cargo build --release` and  then added the `target` directory to my system path. Not a great solution.

Xcodo, 2021-07-19:
The VSCode extension downloads the executable from the [releases section](https://github.com/VHDL-LS/rust_hdl/releases) of this repo. See [getLatestServerVersion()](https://github.com/VHDL-LS/rust_hdl_vscode/blob/8e7f668670d1f262049eff76c46a50b703cd3920/client/src/extension.ts#L285) Would the same thing work for you?

SethGower, 2021-07-19:
I think I actually misspoke. It isn't a problem with installing the language server, but installing `vhdl_lang`. See issue #113. 

I don't think this should be the responsibility of the extension to download it. Maybe it should _be able to_, but not have to. If you have them in your path already, then that should be sufficient.

jsanchez19, 2021-09-03:
@pidgeon777 
> I confirm this works.

Same here but I had to make some changes, which may not be apparent at first glance. Specifically, the "global" lspconfig is defined with the first line below. Then I replaced `configs`  with `require 'lspconfig/configs'`, and finally replaced the global `util`  with `lspconfig.util`. 

```lua
local lspconfig = require('lspconfig')

if not lspconfig.rust_hdl then
  require'lspconfig/configs'.rust_hdl = {
    default_config = {
      cmd = {"vhdl_ls"};
      filetypes = { "vhdl" };
      root_dir = function(fname)
        return lspconfig.util.root_pattern('vhdl_ls.toml')(fname)
      end;
      settings = {};
    };
  }
end
```

I also have the following line
```lua
lspconfig["rust_hdl"].setup { on_attach = on_attach }

```

jsanchez19, 2021-09-03:
@SethGower  - I didn't find that the `vhdl_libraries` directory is needed for running `vhdl_ls`, is that a `vhdl_lang` dependency only? I

SethGower, 2021-09-09:
Yes, I believe it is only a requirement for `vhdl_lang`, but since `vhdl_ls` depends on `vhdl_lang`, it doesn't work. Unless I am missing something.

jsanchez19, 2021-09-09:
yes perhaps as I'm able to run `vhdl_ls` without issues.

Ahmad-Zaklouta, 2021-09-22:
> Late to the game, but figured I'd share this. With neovim's built-in LSP, using `neovim/nvim-lspconfig` this is the custom server declaration I made, and it works.
>
> ```lua
> if not lspconfig.rust_hdl [...]

I tried this in my init.vim 
```
:lua << EOF
local nvimlspconfig = require('lspconfig')
local configs = require('lspconfig/configs')
local util = require('lspconfig/util')
EOF

:lua << EOF
if not nvimlspconfig.rust_hdl then
  configs.rust_hdl = {
  | default_config = {
  | ¦ cmd = {"vhdl_ls"};
  | ¦ filetypes = { "vhdl" };
  | ¦ root_dir = function(fname)
  | ¦ ┆ return util.root_pattern('vhdl_ls.toml')(fname)
  | ¦ end;
  | ¦ settings = {};
  | };
  }
end
EOF
```
but I'm getting 
> E5108: Error executing lua [string ":lua"]:1: attempt to index global 'nvimlspconfig' (a nil value)

jsanchez19, 2021-09-22:
try putting both of those sections in a single `lua << EOF ....EOF` section, like this:
```lua
lua << EOF
local nvimlspconfig = require('lspconfig')
local configs = require('lspconfig/configs')
local util = require('lspconfig/util')

if not nvimlspconfig.rust_hdl then
  configs.rust_hdl = {
  | default_config = {
  | ¦ cmd = {"vhdl_ls"};
  | ¦ filetypes = { "vhdl" };
  | ¦ root_dir = function(fname)
  | ¦ ┆ return util.root_pattern('vhdl_ls.toml')(fname)
  | ¦ end;
  | ¦ settings = {};
  | };
  }
end
EOF
```

Ahmad-Zaklouta, 2021-09-22:
> try putting both of those sections in a single `lua << EOF ....EOF` section, like this:
>
> ```lua
> lua << EOF
> local nvimlspconfig = require('lspconfig')
> local configs = require('lspconfig/configs') [...]

Thanks, that worked.

Guiltybyte, 2021-12-21:
> try putting both of those sections in a single `lua << EOF ....EOF` section, like this:
>
> ```lua
> lua << EOF
> local nvimlspconfig = require('lspconfig')
> local configs = require('lspconfig/configs') [...]

I'm having issues trying to get this working, lsp-config gives the following error after I've added this to my init.vim:
```
[lspconfig] Cannot access configuration for rust_hdl. Ensure this server is listed in `server_configurations.md` or added as a custom server.
```
I cannot find how I am supposed to add rust_hdl as a custom server however, if anyone can provide any insight on this it would be much appreciated.

ttonnellier, 2021-12-23:
Hi,

Following [this issue found on the lspconfig repo](https://github.com/neovim/nvim-lspconfig/issues/1494), I managed to fix my configuration by using:

```lua
local lspconfig = require 'lspconfig'
local configs = require 'lspconfig.configs'

-- Manual add rust_hdl server
if not configs.rust_hdl then
  configs.rust_hdl = {
    default_config = {
      cmd = {'vhdl_ls'};
      filetypes = { "vhdl" };
      root_dir = function(fname)
        return lspconfig.util.root_pattern('vhdl_ls.toml')(fname)
      end;
      settings = {};
    };
  }
end
```

I hope this helps,
Best

rafaelnp, 2022-01-26:
@ttonnellier thanks, I got the basics are working. 

[image]

kraigher, 2022-12-07:
Sounds like this is done, closing

closed 2022-12-07 by kraigher