VHDL-LS/rust_hdl #20: AST to VHDL

benreynwar, 2018-12-07:
Implement conversion of the AST back into a VHDL text file.

There are a number of possible uses for having the ability to go back from the AST to VHDL.
* Debugging of parsing.  Do VHDL -> AST -> VHDL and check it doesn't effect simulation results.
* Implement an obfuscator by changing the symbol names, and then writing the VHDL out.
* Implement a code formatter like rustfmt for VHDL.
* VHDL2008 to well-supported VHDL transformations.  Do transformations on AST, and then write out VHDL.

But mostly I thought it was a reasonably independent project that I could work on and not get in your way too much :).

kraigher, 2018-12-07:
Yes writing out the VHDL is an independent area to work on that is not currently occupied.
I think this functionality should be located as a `formatter` submodule in the vhdl_parser crate.

Currently the AST does not contain comments so the formatter could not be used for human readable source code or the language server. There is a work package of adding comments to the AST that also open. This would also enable document generators that will want access to the comments.

One end goal of the formatter would be to be able to use it in the language server to implement the formatting request and in this case comments cannot be lost.

kraigher, 2018-12-07:
> In the language server, the transforms for each file could be triggered by the onSave event but there would also need to be some way to make a "clean" build for the initial pass. The client would also need to be able to configure the output directory for generated files and from a performance point of view be able to enable or disable the different transforms.

Would such transforms really need to be part of the language server? In my mental model they would just be a step you run before simulation/building.

> What do you think about having traits for post processing and a listener or visitor for the AST? Main goal would be to make it easier to implement and integrate additional features.

Yes a trait/utility for visiting the entire AST is probably useful for many applications. 

Regarding the order of implementation I would imagine something line this:

1. Implement a formatter for the current AST without comments.
2. Add comments to the AST and update the formatter
3. Implement AST transforms (probably needs semantic analysis and symbol binding which is not finished yet)

Bochlin, 2018-12-07:
I don't think that all transforms need to be part of the language server but there are some good use cases where it would be required.
- Block diagram view
- State machine view 
- Real time update of documentation (e.g. see how the html docs look without having to run a build)

Doing transformations to VHDL-93 at save time in the IDE would enable a faster build since the generated code is always up to date. Doing it as part of the build process and outside of the language server would require a full parse + code generation of the entire project. If the VHDL-93 code is generated at save time its already there when you need to build and I don't think that it would be noticeable hit to performance for the user (might even be possible to run the generator in a separate thread). The [XText](https://www.eclipse.org/Xtext/) framework uses this approach for transforming a custom DSL into some intermediate representation.

kraigher, 2018-12-08:
Regarding comments in the AST I created a new issue #22 where I wrote down my thoughts on the subject.

mbrobbel, 2018-12-12:
I'm working on this.

benreynwar, 2018-12-12:
I've finished the tokenizing in #23.  Haven't started moving them from the tokens to the AST yet.

benreynwar, 2018-12-12:
Just realized this is the AST to VHDL, not the comments to AST issue.