VHDL-LS/rust_hdl #25: Recovery in declarative parts

kraigher, 2018-12-10:
# Terminology
A declarative part contains declarations. For the purpose of recovery we divide declarations into two categories. 
- Compound - may contain nested declarative parts, example subprograms, protected type
- Simple - does not contain nested declarative parts, example object declaration alias

# Recovering from error in a simple declaration
Simply scanning for a synchronization token is enough.

## Examples
### Recover by synchronizing with second constant keyword:
```vhdl
constant bad :
variable good : natural;
```  
### Recover by synchronizing with the end of the declarative region 
```vhdl
constant bad :
begin
```  
```vhdl
constant bad :
end
```  

# Recovering from error in a compound declaration
A compound declaration may contain nested declarations and thus recovery cannot simply search for a synchronization token unless we risk synchronizing with a nested declaration.

## Examples
### Error in nested declarative part
```vhdl
-- parent region

procedure proc is
  procedure nested is
  begin
     if -- Syntax error
  end;

  -- We cannot synchronize with this if the above syntax error 
  -- is propagated all the way back to the parent region
  variable good : natural;
begin
end;
```

FenrisTC, 2018-12-10:
This problem would go away, if we would do proper cleanup/recovery of all items in the declarative region, right?

kraigher, 2018-12-10:
Yes the problem is handled if each compound declarative part did its own recovery it would not need to return Err unless there was some unrecoverable error within. 

It would mean that the parent should not try to recover from unrecoverable errors in compound children but only in simple children. If the child compound declaration could not recover the parent will not be able to.

FenrisTC, 2018-12-10:
So the way to go would be to temporarily remove recoverabilty for elements that have no implementation for recovery yet.
I'll update #24 to exclude compound cases for now then.

kraigher, 2018-12-10:
Well we would never try to recover from `Err` from the compound declarative items in the parent so it would not be a temporary solution. Since the `parse_declarative_part` function is called recursively it will work when the children recovered and thus return `Ok`.

FenrisTC, 2018-12-10:
True enough.
I thought about cases in which a declarative item could contain a `Begin` or `End` token but which could not be recovered from.
But when I think about it now, I think we can recover from all cases which are delimited by `Begin` or `End`.