VHDL-LS/rust_hdl #31: User-defined attributes not supported

Xcodo, 2019-03-27:
I've been trying rust_hdl with my projects which are mostly targeting Xilinx FPGAs with the Vivado tools. It's looking very, very useful (thanks for the hard work!), but I've got a bit of a problem with the parsing.

The tool throws a lot of errors with the message
> Expected 'entity', 'architecture', 'configuration', 'package', 'signal', 'variable', 'procedure' or 'function'

wherever user-defined attributes are used. Typically there are only a few of these in my own VHDL files, but when I include the Xilinx component libraries I have thousands of errors.

The problem also prevents other errors being thrown (related to #25 I guess). In the sample below there is an extra comma on the instantiation which is not flagged.

Here's a sample which shows the problem. It happens whether or not there's a `vhdl_ls.toml` included.

`sample.vhd`:
```vhdl
entity sample1 is
	port (
		CLK : in std_logic;
		nCLK : out std_logic
	);
end entity;

architecture a of sample1 is

	component sample2 is
		port (
			CLK : in std_logic;
			nCLK : out std_logic
		);
	end component sample2;

	attribute BOX_TYPE : string;
	attribute BOX_TYPE of sample2 : component is "user_black_box";

begin
	Inst_sample2 : sample2
		port map	(
			CLK <= CLK,
			nCLK <= nCLK,
		);
end a;

entity sample2 is
	port (
		CLK : in std_logic;
		nCLK : out std_logic
	);
end entity;

architecture a of sample2 is
	signal s_nclk : std_logic;
begin
	s_nclk <= not CLK;
	nCLK <= s_nclk;
end a;
```

kraigher, 2019-03-27:
Should be fixed now. The component entity class of the attribute specification was missing. Now your sample code works except it contains a syntax error with a comma at the last port mapping in Inst_sample2.

Xcodo, 2019-03-28:
Thanks for that. I tried it on the libraries today. There's another couple of EntityClass needed to successfully parse the common standard libraries Xilinx uses.

Here's an updated sample with examples of the other types I encountered:
```vhdl
entity sample1 is
  port (
    CLK : in std_logic;
    nCLK : out std_logic
  );
end entity;

architecture a of sample1 is

  --From numeric_std.vhd line:1302
  function "abs" (ARG: SIGNED) return SIGNED is
    constant ARG_LEFT: INTEGER := ARG'LENGTH-1;
    alias XARG: SIGNED(ARG_LEFT downto 0) is ARG;
    variable RESULT: SIGNED(ARG_LEFT downto 0);
    -- Exemplar synthesis directives :
    attribute IS_SIGNED of ARG:constant is TRUE ;
    attribute SYNTHESIS_RETURN of RESULT:variable is "ABS" ;
  begin
    if ARG'LENGTH < 1 then return NAS;
    end if;
    RESULT := TO_01(XARG, 'X');
    if (RESULT(RESULT'LEFT)='X') then return RESULT;
    end if;
    if RESULT(RESULT'LEFT) = '1' then
      RESULT := -RESULT;
    end if;
    return RESULT;
  end "abs";
  
  --From std_logic_1164.vhd:42
  TYPE std_ulogic IS ( 'U',  -- Uninitialized
                       'X',  -- Forcing  Unknown
                       '0',  -- Forcing  0
                       '1',  -- Forcing  1
                       'Z',  -- High Impedance   
                       'W',  -- Weak     Unknown
                       'L',  -- Weak     0       
                       'H',  -- Weak     1       
                       '-'   -- Don't care
                      );

     ATTRIBUTE logic_type_encoding : string ;
     ATTRIBUTE logic_type_encoding of std_ulogic:type is
                     ('X','X','0','1','Z','X','0','1','X') ;

  component sample2 is
    port (
      CLK : in std_logic;
      nCLK : out std_logic
    );
  end component sample2;

  attribute BOX_TYPE : string;
  attribute BOX_TYPE of sample2 : component is "user_black_box";

begin

  Inst_sample2 : sample2
    port map	(
      CLK <= CLK,
      nCLK <= nCLK
    );
end a;

entity sample2 is
  port (
    CLK : in std_logic;
    nCLK : out std_logic
  );
end entity;

architecture a of sample2 is
  signal s_nclk : std_logic;
begin

  s_nclk <= not CLK;
  nCLK <= s_nclk;

end a;
```

I've no experience in Rust, but based on your change last night I went ahead and added the additional entity classes and re-tested. I can now parse my whole project including the libraries. I'll add a pull request now.

PR #32 (2019-03-28): Add constant and type entity class in attribute specifications
Fixes #31.

Happy to assign copyright to the project owner.
closed 2019-03-28 by kraigher