VHDL-LS/rust_hdl #34: user-defined attributes for some entity classes are not supported

rroohhh, 2019-04-28:
Similar to #31 `label`, `subtype`, `literal`, `units`,`group`,`file`,`property` and `sequence` is also not supported for user-defined attributes.

See IEEE Std 1076-2008 7.2 (p. 95):
```
attribute_specification ::=
    attribute attribute_designator of entity_specification is expression ;

entity_specification ::=
    entity_name_list : entity_class

entity_class ::=
    entity
    | architecture
    | configuration
    | procedure
    | function
    | package
    | type
    | subtype
    | constant
    | signal
    | variable
    | component
    | label
    | literal
    | units
    | group
    | file
    | property
    | sequence
```

kraigher, 2019-04-28:
Which ones are you missing?

rroohhh, 2019-04-28:
`label` is the one I am hitting right now

kraigher, 2019-04-29:
I added the label entity class.

rroohhh, 2019-04-29:
Awesome, together with #35 fixed it runs through my codebase without errors.

Should I leave the issue open, to track the other missing entity classes?

kraigher, 2019-04-30:
yes leave it open

closed 2020-01-04 by kraigher