VHDL-LS/rust_hdl #36: vhdl_ls and vhdl_parser are not able to parse standard package generated by ghdl --disp-standard

maehne, 2019-04-30:
I tried to set up `vhdl_ls` for my VHDL project, for which I use `ghdl` for simulation. To fully profit of `vhdl_ls`, I also created a `vhdl_ls.toml` file referencing my project sources as well as the sources of the used packages from libraries `std` and `ieee`. I found the sources of the standard packages in my GHDL installation below `$PREFIX/lib/ghdl/src/`. I only missed the source of `std.standard`. Using the `standard.vhd` file from your `example_project`, I was able to parse my project sources and use `vhdl_ls` in my Emacs. However, your `standard.vhd`is for VHDL'08 and my project is still restricted to VHDL'93. So, I looked wether `ghdl` could provide the source of `std.standard` (cf. to [ghdl issue #802](https://github.com/ghdl/ghdl/issues/802)). @tgingold pointed me to use `ghdl --disp-standard --std=93` to generate the VHDL source of `std.standard`.

Unfortunately, `vhdl_parser` and `vhdl_ls` fail to parse the generated VHDL source. `vhdl_parser` generates parsing errors on at least the definition of type `UNIVERSAL_INTEGER` (Integer too large for 64-bits signed), `real`(invalid float literal due to use of `+` in exponent), and `time` (same as for `UNIVERSAL_INTEGER`). It would be great if `rust_hdl`'s VHDL parser could be enhanced to parse the `std.standard` package VHDL source generated by `ghdl --disp-standard` for all supported language standards by `ghdl` (VHDL'87, VHDL'93, VHDL'08).

To reproduce the issue, build `ghdl` and `rust_hdl` from their respective master branches. Then, generate the `std.standard` package VHDL source and try to parse it using `vhdl_parser` of `rust_hdl`:

```sh
$ ghdl --disp-standard --std=87 > standard.v87
$ ghdl --disp-standard --std=87 > standard.v93
$ ghdl --disp-standard --std=87 > standard.v08
$ vhdl_parser standard.v87
$ vhdl_parser standard.v93
$ vhdl_parser standard.v08
```

For example, this is the output for `standard.v93`:

```
$ vhdl_parser standard.v93
Error when parsing standard.v93
error: Integer too large for 64-bits signed
    --> standard.v93:103
     |
101  |    -- function "<=" (<anonymous>: severity_level;
102  |    --                <anonymous>: severity_level) return boolean;
103 -->   type UNIVERSAL_INTEGER is range -9223372036854775808 to 9223372036854775807;
     |                                     ~~~~~~~~~~~~~~~~~~~
104  |    -- function "=" (<anonymous>: <UNIVERSAL_INTEGER>;
105  |    --               <anonymous>: <UNIVERSAL_INTEGER>) return boolean;

error: Expected 'library', 'use', 'context', 'entity', 'architecture', 'configuration' or 'package'
    --> standard.v93:103
     |
101  |    -- function "<=" (<anonymous>: severity_level;
102  |    --                <anonymous>: severity_level) return boolean;
103 -->   type UNIVERSAL_INTEGER is range -9223372036854775808 to 9223372036854775807;
     |                                                         ~~
104  |    -- function "=" (<anonymous>: <UNIVERSAL_INTEGER>;
105  |    --               <anonymous>: <UNIVERSAL_INTEGER>) return boolean;

Summary:
Found no warnings
BAD: Found errors in 1 files
```

maehne, 2019-04-30:
I have further analysed, which parts of the `std.standard` package VHDL source generated by `ghdl --disp-standard` poses a problem to `rust_hdl`. The parsing error reported for the type definition of `UNIVERSAL_INTEGER` and the `time` data type are caused by the value of the lower range bound:

```vhdl
type UNIVERSAL_INTEGER is range -9223372036854775808 to 9223372036854775807;
type time is range -9223372036854775808  to 9223372036854775807  units
  fs;
  -- ...  
end units;
```
The range for these two integer types corresponds exactly to the range of signed integers, which you can represent using 64 bits in two's complement representation. If `-9223372036854775808` is changed to `-9223372036854775807` (i.e., one added), `rust_hdl` accepts the range. Therefore, I suspect that the parser of `rust_hdl` has a bug. Probably, it parses the sign and absolute value separately and correctly detects that it cannot represent the positive value `9223372036854775808` with 64 bits signed integer.

For the type definitions of `UNIVERSAL_REAL` and `real`, `rust_hdl` does not accept the use of `+` in the exponent part of real values:
```vhdl
type UNIVERSAL_REAL is range -1.79769313486232E+308 to 1.79769313486232E+308;
type real is range -1.79769313486232E+308 to 1.79769313486232E+308;
```
Changing these definitions to
```vhdl
type UNIVERSAL_REAL is range -1.79769313486232E308 to 1.79769313486232E308;
type real is range -1.79769313486232E308 to 1.79769313486232E308;
```
causes `rust_hdl` to accept both type definitions. This means that `rust_hdl`'s VHDL parser needs to get enhanced to accept `+` in the exponent.

Finally, `rust_hdl` found real syntax errors in the sub-units definition of the `time` physical data type. `ghdl --disp-standard` generates the following definition of `time`:
```vhdl
  type time is range -9223372036854775808  to 9223372036854775807  units
    fs;
    ps = 1000 ;
    ns = 1000 ;
    us = 1000 ;
    ms = 1000 ;
    sec = 1000 ;
    min = 60 ;
    hr = 60 ;
  end units;
```
As you may notice, `ghdl` omits the necessary units after the scaling factor values. It should actually look like:
```vhdl
  type time is range -9223372036854775808  to 9223372036854775807  units
    fs;
    ps = 1000 fs;
    ns = 1000 ps;
    us = 1000 ns;
    ms = 1000 us;
    sec = 1000 ms;
    min = 60 sec;
    hr = 60 min;
  end units;
```
I will open an issue against `ghdl`, so that it gets fixed.

kraigher, 2019-04-30:
Your 2 problems should have been fixed now.

maehne, 2019-04-30:
Thanks for addressing so quickly these two issues! I just tried to test it on my laptop runing Debian Buster, which provides `rustc 1.32.0`. Unfortunately, I am running now into a compilation error of `rust_hdl`:
```
$ cd rust_hdl
$ git pull
$ cargo update
    Updating crates.io index
    Updating libc v0.2.51 -> v0.2.53
    Updating num-derive v0.2.4 -> v0.2.5
      Adding numtoa v0.1.0
    Updating proc-macro2 v0.4.27 -> v0.4.29
    Updating syn v0.15.31 -> v0.15.33
    Updating termion v1.5.1 -> v1.5.2
$ cargo build --release
  Downloaded proc-macro2 v0.4.29
  Downloaded syn v0.15.33
  Downloaded libc v0.2.53
  Downloaded num-derive v0.2.5
   Compiling proc-macro2 v0.4.29
   Compiling libc v0.2.53
   Compiling syn v0.15.33
   Compiling num-derive v0.2.5
   Compiling atty v0.2.11
   Compiling clap v2.33.0
   Compiling env_logger v0.6.1
   Compiling quote v0.6.12
   Compiling vhdl_parser v0.4.0 (/home/mht1/Build/rust_hdl/vhdl_parser)
   Compiling serde_derive v1.0.90
error[E0658]: use of unstable library feature 'no_panic_pow' (see issue #48320)
   --> vhdl_parser/src/tokenizer.rs:664:10
    |
664 |         .checked_pow(exp)
    |          ^^^^^^^^^^^

error: aborting due to previous error

For more information about this error, try `rustc --explain E0658`.
error: Could not compile `vhdl_parser`.
warning: build failed, waiting for other jobs to finish...
error: build failed
```
Do you have any hint how to fix it?

rroohhh, 2019-04-30:
You'll need atleast rust `1.34.0` for `checked_pow` in stable, so updating your rust installation should fix that.

maehne, 2019-04-30:
Yes, but Debian Buster will ship with `rustc 1.32.0` and until before the fix for this issue, `rust_hdl` was just building fine with that version. It would be a pity to further complicate the building of this useful tool by requiring users to update their rust compiler above the version their recent Linux distribution provides.

kraigher, 2019-05-01:
We can remove the use of checked pow for now

kraigher, 2019-05-01:
I removed the use of checked_pow for now.

maehne, 2019-05-01:
Thanks for removing for now the use of `checked_pow`! `rust_hdl` still has issues with parsing real literals that use a capital `E` for marking the beginning of the exponent, e.g., `2.5E+3` or `-1.79769313486232E+308`.

maehne, 2019-05-01:
Parsing the `std.standard` VHDL source code generated by `ghdl` using the `vhdl_parser` that includes PR #37 will only report the syntax error concerning the sub-units in the `time` data type definition.

closed 2020-01-04 by kraigher