VHDL-LS/rust_hdl #38: lsp-define-stdio-client deprecated

chrbirks, 2019-05-16:
Hi
It seems that lsp-define-stdio-client is deprecated and unless I'm missing something has been completely removed. I was trying to follow your example for lsp-mode in Emacs but got an error that lsp-define-stdio-client cannot be found.
I'm using Emacs 26.2 and latest version of the elp-mode package.

kraigher, 2019-05-17:
Maybe you can ask the elp-mode maintainers for the new way to do this?

chrbirks, 2019-05-17:
After a bit of trial and error I arrived at something that is able to start the vhdl_ls server and connect to it. Although I haven't figured out how to specify the toml file as input.
```
(require 'lsp-mode)
  (lsp-register-client
   (make-lsp-client :new-connection (lsp-stdio-connection "<PATH_TO_RUST_HDL>/target/release/vhdl_ls")
                    :major-modes '(vhdl-mode)
                    :server-id 'vhdl-lsp))
  (add-to-list 'lsp-language-id-configuration '(vhdl-mode . "vhdl-mode"))
  (add-hook 'vhdl-mode-hook #'lsp)
```

kraigher, 2019-12-03:
I updated the readme with the new method.

closed 2019-12-03 by kraigher