VHDL-LS/rust_hdl #41: LSP should handle config file changed and/or created after startup

suoto, 2019-11-17:
Currently Rust HDL looks for the config file when the LSP is started, so any changes to it (for example adding files, fixing libraries, etc) require the server/editor to be restarted.

Ideally, I think restarting or refreshing the server should happen transparently to the user, so that the process of changing and fixing the config file is faster.

kraigher, 2019-11-17:
This is a good idea. It will be more complex if https://github.com/kraigher/rust_hdl/issues/40 is implemented since the `vhdl_ls.toml` file can be still intact but more files are added which match the wildcard.

suoto, 2019-11-17:
Yeah, I think evaluating wildcards frequently brings little benefit and a lot of issues (like picking up temporary files). The LSP specifies a way to run custom commands on the server, maybe the user explicitly requesting a full refresh is good enough.

Bochlin, 2019-11-17:
I would argue that not requiring the user to update a configuration file every time a file is added/removed is of high value. Having to list every single file manually is in my opinion error prone and for large projects it would be hard to get an overview of the actual project (see the vhdl_ls.toml file in the example code). As a user if an IDE I would not expect to be required to micro-manage files. 

While I don't really see the big issue with the server picking up temporary files (that's the behavior I would expect from an IDE), I do agree that evaluating wildcards frequently brings little benefit as it is unlikely that a user is constantly adding or removing files.

I think that being able to send a command to the server to update would be a good enough solution, then either the user can request a refresh or the client can request a refresh when a file is added or removed from the client GUI. On the other hand, running a background thread once every ~1 s to determine if a refresh is required could be an option in my opinion.

kraigher, 2019-11-17:
I think it is a given that we should support wildcards in the config file. I have always intended to do it but not come to it yet. 

The question is just how wildcards would interact with auto-refresh of the project. If we should support auto-refresh it will become complex with wildcards that need to be periodically monitored or if the OS supports that files can be monitored. A much less complex solution is just to support a "refresh" command that the user manually has to send when he has updated the workspace. Such as checking out a new branch or added a new file himself. The solution today would be to just restart the language server but maybe that is not convenient.

I know the Language Server Protocol supports the server requesting the client to watch for changes in the workspace. Maybe this mechanism can be utilized somehow.

Bochlin, 2019-11-17:
Restarting the server is a working solution, in VSCode atleast it is possible to register a FileSystemWatcher which can react to changes in vhdl_ls.toml and a command could be added for manual refresh.

I think that a refresh command is a perfectly viable solution and letting the client/user take responsibility for requesting refreshes. It feels cleaner than just restarting the server and is also the simplest solution while still providing most of the value by enabling wildcard usage. 

Auto refresh from the server would still be nice to have but maybe that is not the most important feature for now.

umarcor, 2019-11-17:
IMHO, both solutions should be applied: rust_hdl can watch for changes in the config file only and the config file can support wildcards. The *manual* (say FileSystemWatcher) refresh command will be used when files are changed without changing the config file. If changing the branch of the repo modifies the config file, it will be automatically refreshed.

HenningHolmDE, 2020-08-18:
I'm currently working on implementing a reload of the project when the config file changes (using an initial `client/registerCapability` with `workspace/didChangeWatchedFiles` and reacting to `workspace/didChangeWatchedFiles` notifications). The corresponding PR should be ready in the next days.

PR #98 (2020-08-25): Reload design when workspace configuration changes
If supported by the client, the server registers for changes to the configuration file using client/registerCapability.
When the configuration file changes, the client send a notification workspace/didChangeWatchedFiles to the server. The configuration is then loaded again and project configuration is updated:
The design state is reset, new files are added and parsed. As required for incremental document updates, source files that are already loaded are parsed from in-memory sources.

This closes #41
closed 2020-08-25 by kraigher