VHDL-LS/rust_hdl #43: Add graceful handling of non-latin-1 characters

Bochlin, 2019-12-03:
Currently the parser/LSP panics when it encounters a non latin-1 character:

`thread 'main' panicked at 'Source was not legal latin-1: Custom { kind: Other, error: "Invalid latin-1 character � in -- ...`

VSCode will attempt to restart the server leading to a new crash and after 5 crashes within 3 minutes VCode will no longer try to start the server. To start the server again, VSCode must be restarted.

It would be better if an error was emitted for the location of the non valid character. The parser can then either stop parsing the file or try to recover.

kraigher, 2019-12-03:
Yes it should handle it more gracefully. The parser does handle it gracefully but in this case the error occurs before reaching the parser when receiving the utf-16 encoded bytes from the LSP message. I will see if it can be resolved in a better way.

Bochlin, 2019-12-03:
Root cause is that VSCode assumes that files are UTF-8,. I have found a way to associate languages with file encoding which should solve most cases. Graceful handling of wrong encoding would still be nice, but it is not critical.

Bochlin, 2019-12-03:
Within comments, chars from any character set are allowed.

> NOTE 2—Comments may contain characters that, according to 15.2, are non-printing characters. Implementations may interpret the characters of a comment as members of ISO/IEC 8859-1:1998, or of any other character set; for example, an implementation may interpret multiple consecutive characters within a comment as single characters of a multi-byte character set.

kraigher, 2019-12-03:
I would have to re-architect a bit to handle non latin-1 in comments. For now I at least added a diagnostic message for non-utf-8 character encodings.

kraigher, 2019-12-03:
Also the LSP specifies that all text should be sent as unicode so the errors are only given when a unicode character is not a valid latin-1 character. The server converts the utf-8 to latin1 before giving it to the language server and it is this layer that the error can occur. I would have to handle the conversion on the fly deeper inside the parsers to be able to handle arbitrary bytes within the comments.

Bochlin, 2019-12-03:
I don't think that it is common at all to have non latin-1 in comments so I don't really see any reason to put much effort into it now. A diagnostic message is good enough.

Bochlin, 2019-12-03:
You can close this issue unless you plan anything else related.

kraigher, 2019-12-03:
Yes I also think it is good enough for now.

closed 2019-12-03 by kraigher