VHDL-LS/rust_hdl #46: Tests fail on Windows

Xcodo, 2019-12-06:
Tests for both language server and parser fail on Windows. I don't think any of them look serious, it's mostly just directory separator conventions.
### vhdl_ls
```
rust_hdl>cargo test --package vhdl_ls
    Finished dev [unoptimized + debuginfo] target(s) in 1.68s
     Running target\debug\deps\vhdl_ls-b008e55f06f9c466.exe

running 7 tests
test vhdl_server::tests::initialize ... ok
test vhdl_server::tests::utf8_to_latin1_conversion_gives_error_diagnostic ... ok
test vhdl_server::tests::did_open_with_diagnostics_and_change_without ... ok    
test vhdl_server::tests::did_open_no_diagnostics ... ok
test vhdl_server::tests::initialize_with_config_missing_files ... FAILED        
test vhdl_server::tests::initialize_with_bad_config ... ok
test vhdl_server::tests::initialize_with_config ... FAILED

failures:

---- vhdl_server::tests::initialize_with_config_missing_files stdout ----
thread 'vhdl_server::tests::initialize_with_config_missing_files' panicked at 'Object({"message": String("Loaded workspace root configuration file: C:\\Users\\mark_\\AppData\\Local\\Temp\\.tmpDOjfq4\\vhdl_ls.toml"), "type": Number(4)}) does not contain string "Loaded workspace root configuration file: C:\\Users\\mark_\\AppData\\Local\\Temp\\.tmpDOjfq4\\vhdl_ls.toml"', vhdl_ls\src\rpc_channel.rs:158:25
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace.

---- vhdl_server::tests::initialize_with_config stdout ----
thread 'vhdl_server::tests::initialize_with_config' panicked at 'Object({"message": String("Loaded workspace root configuration file: C:\\Users\\mark_\\AppData\\Local\\Temp\\.tmpa9KZcR\\vhdl_ls.toml"), "type": Number(4)}) does not contain string "Loaded workspace root configuration file: C:\\Users\\mark_\\AppData\\Local\\Temp\\.tmpa9KZcR\\vhdl_ls.toml"', vhdl_ls\src\rpc_channel.rs:158:25

failures:
    vhdl_server::tests::initialize_with_config
    vhdl_server::tests::initialize_with_config_missing_files

test result: FAILED. 5 passed; 2 failed; 0 ignored; 0 measured; 0 filtered out
```
### vhdl_parser
```
rust_hdl>cargo test --package vhdl_parser
    Finished dev [unoptimized + debuginfo] target(s) in 0.10s  
     Running target\debug\deps\vhdl_parser-2639f44ec7d3e3aa.exe

running 525 tests
```
ommited test list
```
failures:

---- config::tests::test_warning_on_emtpy_glob_pattern stdout ----
thread 'config::tests::test_warning_on_emtpy_glob_pattern' panicked at 'assertion failed: `(left == right)`

Diff < left / right > :
 [
     Message {
         message_type: Warning,
<        message: "Pattern \'parent_folder\\missing*.vhd\' did not match any file",
>        message: "Pattern \'parent_folder/missing*.vhd\' did not match any file",
     },
 ]

', vhdl_parser\src\config.rs:420:9

---- config::tests::test_append_config stdout ----
thread 'config::tests::test_append_config' panicked at 'assertion failed: `(left == right)`

Diff < left / right > :
 Config {
     libraries: {
         "lib1": LibraryConfig {
             name: "lib1",
             files: [
<                "parent_folder0\\pkg1.vhd",
>                "parent_folder0/pkg1.vhd",
             ],
         },
         "lib3": LibraryConfig {
             name: "lib3",
             files: [
<                "parent_folder1\\pkg3.vhd",
>                "parent_folder1/pkg3.vhd",
             ],
         },
         "lib2": LibraryConfig {
             name: "lib2",
             files: [
<                "parent_folder0\\pkg2.vhd",
<                "parent_folder1\\ent.vhd",
>                "parent_folder0/pkg2.vhd",
>                "parent_folder1/ent.vhd",
             ],
         },
     },
 }

', vhdl_parser\src\config.rs:323:9

---- config::tests::test_warning_on_missing_file stdout ----
thread 'config::tests::test_warning_on_missing_file' panicked at 'assertion failed: `(left == right)`

Diff < left / right > :
 [
     Message {
         message_type: Warning,
<        message: "File parent_folder\\missing.vhd does not exist",
>        message: "File parent_folder/missing.vhd does not exist",
     },
 ]

', vhdl_parser\src\config.rs:344:9

---- config::tests::config_from_str stdout ----
thread 'config::tests::config_from_str' panicked at 'assertion failed: `(left == right)`

Diff < left / right > :
 [
     "C:\\Users\\mark_\\AppData\\Local\\Temp\\.tmphdYH7e\\pkg2.vhd",
>    "\\\\?\\C:\\Users\\mark_\\AppData\\Local\\Temp\\.tmpfc9mko\\absolute.vhd",
 ]

', vhdl_parser\src\config.rs:266:9
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace.

failures:
    config::tests::config_from_str
    config::tests::test_append_config
    config::tests::test_warning_on_emtpy_glob_pattern
    config::tests::test_warning_on_missing_file

test result: FAILED. 521 passed; 4 failed; 0 ignored; 0 measured; 0 filtered out
```

kraigher, 2019-12-06:
Ok now you know what type of operating system I am using. I will fix it in a moment.

Xcodo, 2019-12-06:
Not a problem. That's what CI is for 😉.

kraigher, 2019-12-06:
I fixed the hardcoded unix paths which were the cause of three errors. The rest of the error are something with canonicalize on windows.

kraigher, 2019-12-06:
I believe I fixed all windows errors now, but I have not tested under windows myself.

kraigher, 2019-12-06:
I added an issue about Windows CI: https://github.com/kraigher/rust_hdl/issues/47

Xcodo, 2019-12-06:
Builds now pass (see [CI build](https://travis-ci.org/Xcodo/rust_hdl/builds/621671555)).

closed 2019-12-06 by Xcodo