VHDL-LS/rust_hdl #47: Add GitHub CI for Window and Linux replacing Travis

kraigher, 2019-12-06:
- Replace Travis with GitHub CI.
- Run unit tests on Linux & Windows.

Xcodo, 2019-12-06:
I did a super-simple GitHub Actions cross compilation task [on my fork](https://github.com/Xcodo/rust_hdl/blob/master/.github/workflows/build-test-all.yml). Seems to work.

I can add a pull request if you want to start from there? Or do you want to mirror the more complex Travis setup you have first?

kraigher, 2019-12-06:
Yeah we can add it now and remove travis later when github ci makes it redundant

Bochlin, 2019-12-28:
@Xcodo Wouldn't it make sense to compile and run tests for windows on windows?

Xcodo, 2019-12-28:
Yes it probably would. I started using cross on Travis using `cross` then transferred my config over without looking at the runners hard. I've seen your VSCode setup and was going to go back and look at doing it on Windows. I would also use the maintained cargo actions rather than shelling commands to cargo

Xcodo, 2019-12-28:
Too many ideas, not enough time in front of a PC at the moment!

kraigher, 2019-12-30:
The only thing travis does that the github workflow doesn't is to publish to crates.io and build and publish the docker image.

Xcodo, 2019-12-30:
Both of those will need the keys added to the project secrets store.

PR #54 (2020-01-04): Add workflow to publish to Crates.io
This workflow publishes the crates to Crates.io. It is triggered when new tags are pushed.
I also removed the `.travis.yml` file. This closes #47.

@kraigher you need to add the `CRATES_IO_TOKEN` secret for this to work.
closed 2020-01-04 by kraigher