VHDL-LS/rust_hdl #55: Support adding files to built-in IEEE library

Bochlin, 2020-01-07:
In order to add e.g. VITAL packages to the IEEE library, the entire ieee library must be redefined as any vhdl_ls.toml defining an ieee library with the VITAL files will override the built-in definition. Same problem occurs for the Synopsys libraries, 

- VITAL could probably be supported out of the box using VITAL 2000, I don't think that there are any critical differences for an editor. The license is unclear as the header does not state any copyright or license (GHDL distributes it though).
- Synopsys libraries should be supported out of the box by `vhdl_ls` in order to better support legacy designs. The license states that they can be freely distributed. 

There would still be some corner cases with e.g. Mentor's packages but those could be solved by re-defining ieee.

kraigher, 2020-01-07:
I do not mind accepting a PR adding the libraries to the builtins if they are not restricted for distribution.

Bochlin, 2020-01-07:
I can make a PR. The Synopsys libraries are explicitly allowed to be distributed.  
The VITAL packages are not explicitly restricted and GHDL added them 10 years ago (according to github) but e.g. Sigasi does not distribute them. I have not been able to find anything definitive on how they can be distributed.  

If VITAL is not added to the integrated libraries then adding some kind of append syntax to `vhdl_ls.toml` would be a solution.

kraigher, 2020-01-07:
If GHDL has had them for 10 years I think it can be considered safe to add them.

PR #56 (2020-01-08): Add Synopsys and VITAL 2000 packages
This pull requests adds the Synopsys `std_logic_arith`, `std_logic_unsigned` and `std_logic_signed` packages, and the VITAL 2000 packages.

Resolves #55
closed 2020-01-08 by kraigher