VHDL-LS/rust_hdl #58: Sublime Plugin

nfrancque, 2020-01-09:
I use Sublime as editor and would like to try this out.  Does anyone know if you can/should add sublime package to the main repo, or a separate one?

GHDL has a [https://packagecontrol.io/packages/SublimeLinter-contrib-ghdl](linter) plugin implemented as a separate repository.

kraigher, 2020-01-09:
I do not know anything about sublime but conceptually the only thing the editor needs to know about the server is where the binary is located. VHDL LS communicate with the client using stdio which is the standard and most common method. The editor might need to know about the "workspace root" where the vhdl_ls.toml file is located that contains your library mapping. Editors supporting the LSP protocol typically offer the user to open a workspace folder. You can still run without library mapping but that disables all semantic analys so no find reference or goto declaration. Without library mapping only parsing is done without analysis which gives you the syntax errors only.

It is very easy to build vhdl_ls if you have the rust toolchain installed. We also publish binary packages on github actions for every commit for windows and linux containing the binary as well as the standard and ieee libraries.

kraigher, 2020-01-10:
I realize I might not have answered your main question. Are you asking about adding a sublime plugin to this repository? I think it is better that such plugins each have their own repository that is owned by the people interested in that editor particular editor. Currently the VSCode and Atom plugins are developed in separate repos. Emacs has got support added to the lsp-mode. There is really not so much dependencies between such plugins and the rust_hdl repo. They just need a path to a built-binary.

nfrancque, 2020-01-10:
Okay, in the past I have only used SublimeLinter which is their generic lint-on-save plugin.  That seems less exciting than what you can do.  There is a sublime package that appears to implement LSP and I've asked on their discord what the process of getting it on there would be.  

Otherwise yes would agree it makes sense to keep those repos separate.

kraigher, 2020-01-10:
Yes running a language server is night and day vs. running an external tool on save.
You get errors instantly as you type and also all the semantic code navigation such as find references and goto declaration.

If you just want to try it quickly I would recommend using VSCode. VSCode is super easy to install and then you just install the VHDL LS plugin by searching for it within the integrated VSCode extension browser and then clicking "install".

nfrancque, 2020-01-10:
As a testament to how well things are working - between that comment and now I managed to get VSCode installed and VHDL LS loaded up and working!

I will continue seeing if getting it added to Sublime is doable but very cool that it works so easily.  Thanks for the help.

nfrancque, 2020-01-11:
For posterity in case some else wants to do this:

The package [LSP](https://github.com/sublimelsp/LSP) implements LSP for Sublime.  As of the time of writing this it does not package the language server with it; the user installs it in whatever way necessary.  Sounds like a discussion about that is ongoing.

Once LSP is installed access its settings either through Command Pallete->Preferences: LSP Settings or Packages/User/LSP.sublime-settings in the Sublime config folder.  Replace its contents with the following:

```
{
    "clients":
    {
        "vhdl_ls":
        {
            "command":
            [
                // Replace with your path to the built vhdl_ls executable
                "/path/to/rust_hdl/target/release/vhdl_ls"
            ],
            "enabled": true,
            "languages":
            [
                {
                    "languageId": "vhdl",
                    "scopes":
                    [
                        "source.vhdl"
                    ],
                    "syntaxes":
                    [
                        // Install the package VHDL-Mode
                        // https://packagecontrol.io/packages/VHDL%20Mode
                        "Packages/VHDL Mode/Syntax/VHDL.sublime-syntax"
                    ]
                }
            ]
        }
    }
}
```

I had to kill all Sublime instances and restart before the changes took effect.

Then, do Command Pallete->LSP: Enable Language Server Globally

Have not tested thoroughly but was able to see vhdl_ls output in the right places if I introduced syntax errors.

kraigher, 2020-01-12:
You need to check that also the semantic analysis works as well by doing goto-definition. VHDL LS will can only give you syntax errors if it cannot find the project configuration file (vhdl_ls.toml). The project configuration file should be located in the workspace root. The editor typically needs to let the user choose the workspace folder and sometimes you can add hooks to auto-detect it.

nfrancque, 2020-01-12:
Sorry, should have mentioned that yes I had a vhdl_ls.toml in the project,
and it was picking up types I had defined in other workspace files.
Automatic, no added configuration necessary.

However it looks like the IDE-like features (goto, rename, etc.) are all
grayed out.  Goto definition I imagine must be supported by vhdl_ls,
otherwise it wouldn't be working.  So I suspect that is a Sublime LSP
issue.  What do you think?

kraigher, 2020-01-12:
The server announces its capabilities to the client. We do not support rename yet but goto definition / declaration as well a find references is supported and thus announced and works in other editors.

closed 2022-12-07 by kraigher