VHDL-LS/rust_hdl #69: Precompiled libraries

adrianwn, 2020-01-27:
I would like to implement support for _precompiled libraries_. Precompiled libraries would speed up the initial analysis of a project, because `std` and `ieee` would not have to be parsed and analyzed every time.
My thoughts on the subject:

- `vhdl_lang` should not prescribe a file format, nor should it be involved in locating precompiled libraries on disk.
- `vhdl_lang` should probably not include a reference implementation of the (de)serialization step. `vhdl_ls` is a different matter though, and should probably provide such an implementation. Not sure on that point...
- Since `vhdl_lang` isn't able to locate, load, and deserialize a precompiled library, and programs using `vhdl_lang` should, performance reasons, not be forced to load and deserialize every available library beforehand, the following procedure would make sense:
  1. Create a `Project` and parse the source files.
  2. Call a method `Project::get_missing_libs()` which returns the set of library names which are referenced in the library clauses of any parsed design unit, minus the set of library names which are already known to the `Project` instance.
  3. Load and deserialize the missing libraries. The newly constructed ASTs and `Library` instances are added to the `Project`.
  4. Call `Project::analyse()`. No changes should be needed anywhere in the analysis code.

Regarding the concrete (de)serialization implementation:
- I can see two major use cases for precompiled libraries:
  1. _Black box libraries_: A "closed" library without references to the original source files; provided by, e.g., a simulator vendor. `SrcPos` instances of declarations would probably be stored as some form of "named entity #17259 in library 'xyz'".
  2. _Work in progress libraries_: A library which retains full `SrcPos` data. Typical scenario: when working with command line tools like ModelSim's `vcom` and `vsim`, which incrementally add source files to a working library.
- Do you have any plans regarding the `Reference` type? Currently it uses `SrcPos` as a way to identify declarations, which is okay for a language server, but suboptimal for a simulator or synthesis tool. It's also a bit problematic when serializing a black box library.

kraigher, 2020-01-27:
First of all are you sure it is the right time to add precompiled libraries now? vhdl_lang can already analyze 200 kloc of code in 200 ms. To me it seems like a premature optimization to focus on speed at this point rather than completing other missing functionality. It is also probably hard to make pre-compiled libraries significantly faster than parsing from scratch since they would actually take up more space on disk and thus be slower to load from disk which is a bottle neck. Adding a lot of serialization/de-serialization code now could make it harder to refactor internal data structures which I still see as likely until type checking is complete. I am not ruling it out pre-compiled libraries in the future but it seems now might not be the right time to do it.

Regarding the Reference type it is not planned to be a SrcPos in the future. Ideally it would be a pointer to a NamedEntity. Making it a SrcPos was a shortcut to get goto declaration and find all references to work. 

Regarding NamedEntity I am thinking about making it an enum type with different structs for types, objects etc. Currently I store a NamedEntity in many places where I know it must be a type or object but I cannot express it in the type system. Since I cannot express that something is a type I must thus use `match` to get the value that I already know must be there and put `unreachable!` in the other cases which is not so nice.

adrianwn, 2020-01-27:
> First of all are you sure it is the right time to add precompiled libraries now? [...] Adding a lot of serialization/de-serialization code now could make it harder to refactor internal data structures which I still see as likely until type checking is complete.

Hmm, yes, I see your point. The thing is, for me precompiled libraries are not just interesting from the performance aspect, but even more so because I want to make my simulator compatible to ModelSim [1].
However, I respect your decision and won't push the issue. Once the AST is more or less stable, I would be glad to contribute an implementation of precompiled libraries. For now, re-parsing and -analyzing everything should be sufficient to test and tune my simulation algorithm. I'm really excited to see this project going forward!

[1] I've designed an algorithm for a multi-threaded, event-based simulator. This will be a commercial product, so it would help a lot if my customers could use it as a drop-in replacement for ModelSim and other simulators, without having to adjust their scripts. For this, I need a way to incrementally add source files to a library, i.e., to serialize and deserialize the analyzed AST.

kraigher, 2020-01-27:
I am open to adding serialization/deserialization in the future but today I think it is premature and will slow us down. Once the core features such as type checking is done we can reevaluate. Also we need to discover the best way to structure data for a simulation application.

Interesting that you aim to create a commercial simulator. My advice would be to consider the business case. I think it is extremely hard to make money on it. On the one side you have free tools such as GHDL/Verilator to compete with. On the other side you have commercial tools with decades of polish in speed and debugging capabilities were the development cost has already been repaid. Even if your algorithm was 2x faster than commercial tools you would still struggle to get customers due to trust in correctness and debugging capabilities. Consider also that a lot of IP is encrypted which creates a barrier to entry. For many use cases a single simulator does not need to be parallel because you often run thousands of tests and those can run in parallel resulting in speedup of P. It might still be fun to create a simulator but I sincerely advice against starting a business based on it.

adrianwn, 2020-01-29:
I understand and share your concerns. Those obstacles are real and will be hard to overcome.
However, I believe a faster simulator (by a lot more than 2x, I hope) will be truly desirable. When debugging a testbench, it doesn't help that I could run 50 of them in parallel: For larger designs, you're easily wasting an hour or more for every run.
It's even worse for gate-level simulations. Imagine you had a 10x faster simulator – suddenly, you could actually test interesting stuff because you don't have to wait for a day or two until design startup and initialization is done.
It's quite likely that I fail, but at least I have to try.

> Once the core features such as type checking is done we can reevaluate.

Alright, I'm going to close this issue for now.

closed 2020-01-29 by adrianwn