VHDL-LS/rust_hdl #70: Case statement in a process should auto-insert all states if the select variable is an enumerate

GlenNicholls, 2020-01-28:
When using the case syntax, a nice feature would be to automatically insert all case names. To accomplish this, the parser would need to see that the case variable as shown below is an enumerate, and if so, it would automatically insert all options:

```
type fsm_t is (IDLE, STEP_1, DONE);
signal state : fsm_t := IDLE;

...
process(all)
begin
    case state is -- after user presses tab when entering case select variable, it will auto populate options like below only if parser recognizes signals type as an enumerate
        when IDLE =>

        when STEP_1 =>

        when DONE =>

        when others => null; -- could have option for the user to set state to first option in enumerate for the others case
    end case;
end process;
```

GlenNicholls, 2020-01-28:
Moved here from https://github.com/Bochlin/rust_hdl_vscode/issues/11

kraigher, 2020-01-28:
The vhdl ls language server currently does not provide any snippets or completions at all. We simply have no implementation of it now. 

If you have seen auto complete in of a case statement it was likely done by some VHDL snippet in the client and not in vhdl ls. Thus there is nothing that we can do in the short term to accomodate your request. It is an empty area today and not high on the prio list. 

In the future however we should have this type of functionality in the language server. The LSP protocol defines support for it. The primitive snippet support in a client can only go so far since it does not have semantic information about enums etc. That is why the language server should implement it.

closed 2022-12-07 by kraigher