VHDL-LS/rust_hdl #71: Status of the project.

m-kru, 2020-02-17:
@kraigher Is there anywhere information about what you are working on right now, and what are the plans for _near_ future?

kraigher, 2020-02-17:
The last improvement was adding signatures to subprograms. That is the basis of resolving overloaded subprogram calls which would be the next step. Since I started a new job the activity has been low due to other priorities.

Do you have some specific things in mind?

m-kru, 2020-02-17:
I was just loosely thinking about trying to implement PSL support with some SAT solver. It requires robust semantics parsing. I have not analyzed rust_hdl code yet and I do not know if it is a good point to start. I know that GHDL has partial support for PSL in simulation, however I would prefer writing in Rust than Ada.

kraigher, 2020-02-17:
It is hard to know what is missing without trying.

m-kru, 2020-02-17:
Deep and true.

closed 2020-02-17 by m-kru