VHDL-LS/rust_hdl #79: Crates on https://crates.io/

m-kru, 2020-06-27:
On https://crates.io/ there are two crates [vhdl_parser](https://crates.io/crates/vhdl_parser) and [vhdl_lang](https://crates.io/crates/vhdl_lang). Both point to this repository. However there is no directory named `vhdl_parser`, even when you  grep for  *vhdl_parser* you get nothing. Can you explain how [vhdl_parser](https://crates.io/crates/vhdl_parser) is created and what is the difference between [vhdl_parser](https://crates.io/crates/vhdl_parser) and [vhdl_lang](https://crates.io/crates/vhdl_lang)?

Xcodo, 2020-06-27:
The package was renamed from `vhdl_parser` to `vhdl_lang` in January ([this commit](https://github.com/kraigher/rust_hdl/commit/9ab66c2c2f3ff7aef7a982d6020e895bdda0b3da). From v0.14 onwards the package is called `vhdl_lang`.

m-kru, 2020-06-27:
I wish, this information was on the [vhdl_parser](https://crates.io/crates/vhdl_parser).

closed 2020-06-27 by m-kru

Xcodo, 2020-06-27:
@kraigher could the description of the old `vhdl_parser` crate be updated on `crates.io`?

kraigher, 2020-06-27:
I haven seen a way to remove it or change the description.

Xcodo, 2020-06-27:
Looks like crates can't be removed from the docs.

I wonder if you published a new version whether you could change the description in the manifest to say that is has been superceded?