VHDL-LS/rust_hdl #80: Improve documentation

m-kru, 2020-06-27:
The current documentation is almost non-existing. @kraigher could you please document at least `vhdl_lang` crate public items?
I am trying to use `vhdl_lang` as dependency, but I stuck analyzing the internals. I guess the point of sharing code via crate is a bit different.

kraigher, 2020-06-27:
I am not sure I am willing to invest more time in this project going forward. I am no longer in a job where I work with VHDL so there is less in it for me.

m-kru, 2020-06-27:
I am really sad to hear this. Are you going to accept PRs, or is it better to fork?

kraigher, 2020-06-27:
I can give push rights to people who are interested in taking the project further.

m-kru, 2020-06-27:
Hmm, not sure if it is good idea when the owner of the repository ''abandoned''  the project. Plus you have all the copyrights. I would be interested in taking over.

kraigher, 2020-06-27:
Forking is always an option. But the crates are in my name. Before doing anything radical like changing copyright or ownership I would like to see that there is a sustained interest and work on the project. I am not so keen on just giving away everything to the first person who asks. I would advice you to try and see if you can make a meaningful contribution and see where that leads to.

m-kru, 2020-06-27:
Sure, I understand. I will fork and work on side.

closed 2020-06-27 by m-kru