VHDL-LS/rust_hdl #85: Analysis vs syntax directories.

m-kru, 2020-07-03:
@kraigher would you be so kind to explain, what was the intention of `analysis` directory and how it differs from `syntax` directory? What should be added/looked for in any of them?

kraigher, 2020-07-03:
The syntax directory contains the tokenization and parsing code. This takes VHDL from Bytes to Tokens to Syntax Tree. 

The analysis directory contains the semantic analysis code. The semantic analysis does type checking, overload resolution and resolves names.

kraigher, 2020-07-03:
The analysis part is not complete and only partially does what it is intended to do. Paraing on the other hand is very much solved, except for formatting concerns.

closed 2020-07-03 by m-kru