VHDL-LS/rust_hdl #86: export LSIF file

rgalonso, 2020-07-18:
I'd like to use this program to export the LSIF data to a file so that it can be consumed by GitLab for its ["Code Intelligence"](https://docs.gitlab.com/ee/user/project/code_intelligence.html) feature. This enables hover features, etc. while browsing the code in the web IDE. 

Is generating this LSIF dump currently possible? The statement in the README, "*...has no intended usability at this point*",  makes me think it may not be possible, but I was hoping you could tell me clearly whether it is or not. If not, is that an intended feature in the future? Thanks!

p.s. If this is something that would be a "good first issue", I'd be willing to look into it. I'm interested in working in Rust and have read the documentation for the language, but haven't had an opportunity to work on a real project in that language. It would probably be several months before I have the free time to work on this though.

kraigher, 2021-03-07:
Yes this is probably a good first issue to work on if you want to contribute.

SethGower, 2024-03-22:
I am going to look into implementing this. I am new to the source code, although have some limited experience with Rust. If you have any tips on ideas on how to implement this, feel free to let me know. 

At first perusal of the project, my plan is to investigate using the [`document_symbols(...)`](https://github.com/VHDL-LS/rust_hdl/blob/master/vhdl_lang/src/project.rs#L280) function to see if that will give me a starting point for creating the LSIF dump.

kraigher, 2024-03-23:
Document symbols is a good place to start. It will give you the location where every signal etc (named entity) is defined. 

If you also need every place where a named entity is referenced you would use the Searcher trait to visit every reference in the AST. 

I do not recommend using find all references for every named entity individually since it will traverse the entire AST every time. This is a lot more inefficient than doing it once for all references.

SethGower, 2024-03-24:
Ok, thanks for the tip! I was planning on going the latter method, so thanks. I'll update this thread with progress/questions as I go along.