VHDL-LS/rust_hdl #87: Rationale for Message and MessagePrinter structs.

m-kru, 2020-07-22:
@kraigher would you be so kind to explain what is the rationale for `Message` and `MessagePrinter` structs? Do I understand correctly, that Message has been introduced to abstract the message exchanged with LSP client, but later you wanted to also print some of the messages in the `vhdl_lang` binary, so you have introduced `MessagePrinter`?

kraigher, 2020-07-22:
Yes the point of the trait is to abstract the message consumer which could be either LSP pr stdout.

m-kru, 2020-07-22:
Why no to use MessageType directly from lsp-types
```Rust
pub enum MessageType {
    /// An error message.
    Error = 1,
    /// A warning message.
    Warning = 2,
    /// An information message.
    Info = 3,
    /// A log message.
    Log = 4,
}
```
But introduce custom one
```Rust
pub enum MessageType {
    Error,
    Warning,
    Info,
    Log,
}
```
and then you do the conversion with
```Rust
fn to_lsp_message_type(message_type: &vhdl_lang::MessageType) -> MessageType {
    match message_type {
        vhdl_lang::MessageType::Error => MessageType::Error,
        vhdl_lang::MessageType::Warning => MessageType::Warning,
        vhdl_lang::MessageType::Info => MessageType::Info,
        vhdl_lang::MessageType::Log => MessageType::Log,
    }
}
```

You also decided to do the same with SrcPos. What am I missing?

kraigher, 2020-07-22:
To avoid the non language server part from depending on the language server types.

m-kru, 2020-07-22:
I have reopened to ask one more question. Why not to use only Diagnostic and convert diagnostic to LSP Message? Why Diagnostic and Message are separate entities, they look very similar?

kraigher, 2020-07-23:
One is associated with a source code position and the other one is not. For example a syntax error is a diagnostic and a failure to open a file is a mesage.

closed 2020-07-23 by m-kru