VHDL-LS/rust_hdl #89: Logic behind declaration search.

m-kru, 2020-08-03:
@kraigher I struggle to analyze how the declaration position is determined when LSP client asks for `Go to Declaration`. Everything is clear until `search_reference` method is called on `DesignRoot`
```Rust
    pub fn search_reference(&self, source: &Source, cursor: Position) -> Option<SrcPos> {
        ItemAtCursor::search(self, source, cursor)
    }
```
If I understand correctly, first the item under the cursor is found, what is done with `ItemAtCursor`. By the way, I have temporarily modified `is_inside` function to print some debug info, and it looks like this function returns `true` twice for the same `cursor` and `pos` values during the single `Go to Declaration` procedure.
```Rust
    fn is_inside(&self, pos: &SrcPos) -> bool {
        // cursor is the gap between character cursor and cursor + 1
        // Thus cursor will match character cursor and cursor + 1
        if pos.start() <= self.cursor && self.cursor <= pos.end() {
            error!("{:#?}", self.cursor);
            error!("{:#?}", pos);
            return true
        } else {
            return false
        }
    }
```
Output:
```
[2020-08-03T13:35:53Z ERROR vhdl_lang::ast::search] Position {
    line: 91,
    character: 18,
}
[2020-08-03T13:35:53Z ERROR vhdl_lang::ast::search] SrcPos {
    source: Source {
        source: Source {file_name: "/home/mkru/workspace/smx/rcv_frames_shuffler/rtl/rcv_frames_shuffler.vhd"},
    },
    range: Range {
        start: Position {
            line: 91,
            character: 12,
        },
        end: Position {
            line: 91,
            character: 20,
        },
    },
}
[2020-08-03T13:35:53Z ERROR vhdl_lang::ast::search] Position {
    line: 91,
    character: 18,
}
[2020-08-03T13:35:53Z ERROR vhdl_lang::ast::search] SrcPos {
    source: Source {
        source: Source {file_name: "/home/mkru/workspace/smx/rcv_frames_shuffler/rtl/rcv_frames_shuffler.vhd"},
    },
    range: Range {
        start: Position {
            line: 91,
            character: 12,
        },
        end: Position {
            line: 91,
            character: 20,
        },
    },
}
```
Nevertheless, what I do not understand is what happens when the item under the cursor is already found. How is the actual declaration found? The whole search logic looks a bit distributed to me, and I find it hard to get the overall view.

kraigher, 2020-08-03:
There is a Search trait in search.rs that is used to search the AST in lexicographical order. The AST is traversed until an element is found which has a source position that overlaps the cursor. 

The same Search trait is used to implement the feature that finds all references by traversing the AST and saving all element positions that reference the target.

m-kru, 2020-08-03:
> There is a Search trait in search.rs that is used to search the AST in lexicographical order. The AST is traversed until an element is found which has a source position that overlaps the cursor.

This is the part that I think I understand. But lets assume there is some out port assignment. When I right click it, then `Go to Declaration`, I end up on port declaration in interface declaration. This position is different than what was used on `ItemAtCursor` creation. For example I right click at 
```
Position {
    line: 91,
    character: 15,
}
```
which is used as the cursor position for `ItemAtCursor`, and it takes me to:
```
Position {
    line: 15,
    character: 4,
}
```

m-kru, 2020-08-04:
@kraigher One more question. Why have you decided to implement the search mechanism this way? Based on the cursor position you could fetch the name, calculate the symbol id, and then look for declaration with given id.

kraigher, 2020-08-04:
The method you propose fail to handle nested scopes with identically named variables.

closed 2022-04-07 by m-kru