VHDL-LS/rust_hdl #93: Syntax Highlighting of pure/impure function

FFFFFFFXXXXXXX, 2020-08-13:
"pure function ..." and "impure function ..." don't seem to Syntax highlight properly.
The Problem seems to be that the match for "function" only takes whitespace before "function".
Removing "^\s*" in the file vhdl.tmLanguage fixes the Problem for me.

```
       <key>function_prototype_pattern</key>
        <dict>
            <key>patterns</key>
            <array>
                <dict>
                    <key>begin</key>
                    <string>(?x)
                        # From the beginning of the line
                        ^\s* <-- remove this line
						
                        # The word function $1
                        ((?i:function))\s+
                        ...
```
```
        <key>function_definition_pattern</key>
        <dict>
            <key>patterns</key>
            <array>
                <dict>
                    <key>begin</key>
                    <string>(?x)
                        # From the beginning of the line
                        ^\s* <-- remove this line

                        # The word function $1
                        ((?i:function))\s+
                        ...
```

umarcor, 2020-08-13:
Possible dup pf #92.

FFFFFFFXXXXXXX, 2020-08-13:
#92 can be fixed by adding "pure" to the Keywords in the rust code (tokenizer.rs + some other files).
This one has to be fixed in vhdl.tmLanguage grammar file.
I'm pretty sure those two are not related

closed 2020-08-14 by FFFFFFFXXXXXXX