VHDL-LS/rust_hdl #101: Goto definition/declaration does not work for functions and procedures

HenningHolmDE, 2020-09-08:
While it works fine on most of the definitions, running "Goto definition" on function or procedure calls seems not to be implemented yet.

(Tested on VS Code running current master build 189c98a)

kraigher, 2020-09-08:
Yes, goto definition for overloaded names such as functions and procedures is not implemented. The blocker is overload resolution which in turn requires type inference.

HenningHolmDE, 2020-09-08:
Ah, because in VHDL the correct one of multiple functions with the same name may have to be inferred by the destination type? That's indeed a tough one.

Do you have any plans, concepts or visions on implementing type inference?

kraigher, 2020-09-08:
I have paused my efforts for now. Since I no longer work with VHDL on a daily basis I have less incentive to invest more time into it. That might change in the future or it might not.

HenningHolmDE, 2020-09-08:
What a pity! As someone who works with VHDL on a daily basis, I know how hard it is to find good tooling. So I hope the incentive will return some day!

kraigher, 2021-03-16:
There has now been an improvement in goto defintion and find references for subprograms. It works when there is no overloading and in some cases even for  overloading.

HenningHolmDE, 2021-03-31:
I finally got down to updating to the new release. Goto definition works as expected for non-overloaded subprograms. Thank you very much!