VHDL-LS/rust_hdl #110: [Bug] thread 'main' panicked when running into Neovim v0.5.0 in Windows 10

pidgeon777, 2021-05-25:
In Neovim v0.5.0 for Windows 10 this happens when trying to execute `vhdl_ls.exe` inside of Neovim.

Command:

````
:vhdl_ls.exe
````

Result

````
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: ProtocolError("expected initialize request, got Err(RecvError)")', vhdl_ls\src\stdio_server.rs:80:85
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
````

The strange this is that when executed in Command Prompt, `rust_hdl` works.

How could this bug be solved?

pidgeon777, 2021-05-25:
After setting `RUST_BACKTRACE` to `full`:

````
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: ProtocolError("expected initialize request, got Err(RecvError)")', vhdl_ls\src\stdio_server.rs:80:85
stack backtrace:
   0:     0x7ff786b1c0ae - <unknown>
   1:     0x7ff786b3433b - <unknown>
   2:     0x7ff786b16f38 - <unknown>
   3:     0x7ff786b1ef9d - <unknown>
   4:     0x7ff786b1ea04 - <unknown>
   5:     0x7ff786b1f8ae - <unknown>
   6:     0x7ff786b1f3d1 - <unknown>
   7:     0x7ff786b1ca0f - <unknown>
   8:     0x7ff786b1f329 - <unknown>
   9:     0x7ff786b328f0 - <unknown>
  10:     0x7ff786b32733 - <unknown>
  11:     0x7ff7867a4d4e - <unknown>
  12:     0x7ff7867a13a6 - <unknown>
  13:     0x7ff7867a1686 - <unknown>
  14:     0x7ff7867a148c - <unknown>
  15:     0x7ff786b1fba4 - <unknown>
  16:     0x7ff7867a1417 - <unknown>
  17:     0x7ff786b3bf20 - <unknown>
  18:     0x7ffd11074034 - BaseThreadInitThunk
  19:     0x7ffd111c3691 - RtlUserThreadStart

shell terminato con return-code 101
````

SethGower, 2021-07-19:
You can't just run the binary from Neovim and expect it to work. You have to run it from the LSP client. 

(I know you have gotten that working, since I have seen you commenting elsewhere, figured I'd post this for posterity's sake)

kraigher, 2022-12-07:
The vhdl_ls binary is not supposed to be run by an end user but by a parent editor which communicates with it via stdout/stdin. The error you are seeing is that it cannot read the LSP json rpc message from stdin.

closed 2022-12-07 by kraigher