VHDL-LS/rust_hdl #113: Unable to install from crate and run

SethGower, 2021-07-14:
When installing from the cargo crate, using `cargo install vhdl_lang`, `vhdl_lang -c vhdl_lang.toml --perf` failes with the panic message

```
thread 'main' panicked at 'Couldn't find installed libraries at /home/sgower/.cargo/bin/../../vhdl_libraries/vhdl_ls.toml.', /home/sgower/.cargo/registry/src/github.com-1ecc6299db9ec823/vhdl_lang-0.17.0/src/config.rs:230:13
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```
It seems that it is looking for the `vhdl_libraries` directory that is in the git repo, but isn't part of the actual crate package. 

Test stuff
```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
use ieee.std_logic_unsigned.all;

entity a is
  port(
    i_clk : in std_logic;
    i_rst : in std_logic;
    o_sig : out std_logic
  );
end entity a;
architecture Behavioral of a is
  signal s_1 : std_logica; -- intentional error
begin
  process
  begin
  
  end process;
end Behavioral;
```

```toml
[libraries]
work.files = [
  'test.vhd',
]
```

wrightsg, 2023-01-03:
I just ran into this issue when installing `vhdl_ls` with `cargo install vhdl_ls`. Is this considered an issue and if not, what is the intended way to handle this?

SethGower, 2023-01-04:
> I just ran into this issue when installing `vhdl_ls` with `cargo install vhdl_ls`. Is this considered an issue and if not, what is the intended way to handle this?

Personally in order to use this, I install it by cloning the repo to `~/.local/share/rust_hdl` and then do `cargo build --release` in order to build it. Then I add `~/.local/share/rust_hdl/target/release` to my `$PATH`. It's not a great way, but it works for me.

kraigher, 2023-01-04:
`cargo install` is not really supported since cargo has no concept of installing data folders along with the binary that is needed for the `vhdl_libraries`. The recommended method is to download the latest release from github for your platform or build it locally using cargo build.

teburd, 2023-02-04:
I ran into this as well. I get the issue at hand, but having to do something other than cargo install adds a small but annoying hurdle.

1. could the data be included in the binary itself with some include_file!() style macros?
2. could the data be fetched by vhdl_ls and stored in a user local .cache directory?

I get maybe neither are feasible or no one is willing to add the feature to do so... but, its really hard to beat the convenience of cargo install!

kraigher, 2023-02-04:
Cargo install is just not that useful for end users. You do not want your editor to require you to have cargo installed. 

But yes I could probably include the vhdl files in the binary and extract them somewhere. Extracting to a real file is still necessary so that the user can open the standard.vhd if the for example run goto definition on a standard type.

sazam0, 2023-03-06:
> Personally in order to use this, I install it by cloning the repo to `~/.local/share/rust_hdl` and then do `cargo build --release` in order to build it. Then I add [...]

Since this is working without any issue/complexity, please add this as an (optional) installation instruction in the readme. The other 2 failed attempts (ubuntu 20.04.5, editor: emacs 28 with doom):
1. cargo install: problem with vhdl_libraries directory location
2. download the zip file: requires glibc 22-24, not available by default

kraigher, 2023-03-06:
@sazam0 

1. Regarding install instructions you are welcome to make a PR to improve them.
2. Regarding glibc dependency there is the `musl` target which is 100% static. This is used by the VSCode extension.