VHDL-LS/rust_hdl #115: Erroneous "Opening file that is not part of the project" message

Zalfrin, 2021-07-28:
I'm using VHDL LS from inside vscode running on Windows. When I open a project that is located on my local disk drive (C:), everything works fine. However, if I take the same project and move it to a NAS drive on our network, LS no longer recognizes the VHDL files as belonging to a project.

Maybe network latency is causing something weird to happen?

While investigating, I found a similar bug report for a different extension with a detailed explanation, may or may not be related. https://gitlab.com/gitlab-org/gitlab-vscode-extension/-/issues/290#note_475873929

I'll attach the project I am testing with. It also contains the logs that are produced in vscode with verbose trace running (one for project being locally stored and one for when it is on a  NAS drive).

[vhdl_ls_mre.zip](https://github.com/VHDL-LS/rust_hdl/files/6895200/vhdl_ls_mre.zip)

kraigher, 2023-02-05:
Closed due to lack of activity

Matt-D-Smith, 2023-11-11:
I'm encountering this exact issue as well, and it appears someone on the vscode [extension issues tracker ](https://github.com/VHDL-LS/rust_hdl_vscode/issues/75) is as well. I don't have rust experience, but have a little bit with C, so I tried to take a look at this, but there seems to be no build/debug instructions. I could figure out how to build, but not how to debug it.

kraigher, 2023-11-11:
@Matt-D-Smith you can use `eprintln!()` and it will show up in the `Output` window in VScode if you select `VHDL-LS Trace` in the top left drop-down in the output window.

Matt-D-Smith, 2023-11-11:
Some initial finding, within LibraryConfig() there is a fn as_abspath(). The file_name input to that is of format like ordinary windows file path for a network drive

\\sshfs.r\user@domain\mnt\drive\projects\test\test1.vhd

the code runs as_abspath(&file_path), and puts abs_path in result
abs_path is of format

\\?\UNC\sshfs.r\user@domain\mnt\drive\projects\test\test1.vhd

Elsewhere in the code I don't think this file path format is used, and the windows file path is used instead, which is causing 
"Opening file {} that is not part of the project"
to be triggered, unsure if this file path discrepancy causes issues elsewhere.

I have seen this issue cause VHDL_LS to think there is a duplicate entity of same name when only one entity of that name exists in the project.

kraigher, 2023-11-12:
@Matt-D-Smith could you try the https://github.com/VHDL-LS/rust_hdl/tree/file_path branch? I made a quick hack there to test a hypothesis on how this problem can be fixed.

Matt-D-Smith, 2023-11-12:
Building that branch, I no longer see the "Opening file that is not part of the project" warning with either the workspace file path being the network file path format \\server\share\folder or the network mapped drive format Z:\folder. Seems like that worked!

closed 2023-11-13 by kraigher

Dioswilson, 2024-10-19:
I have a similar issue.
I am running Neovim on linux and when opening a file (directory doesn't matter) and when I open a vhdl file I get the `Opening file <fileName> that is not part of the project` waring.
I created the configuration file called `vhdl_ls.toml` at the same directory the files are at, which is also the current working directory (checked with `:lua print(vim.fn.getcwd())`

commit 77a2b54e (2023-11-13): Ensure paths are consistently canonicalized. Closes #115