VHDL-LS/rust_hdl #119: BEGIN ... END block inside GENERATE statement is not correctly recognized

Hoffti, 2021-08-17:
If a generate statement contains a BEGIN ... END block VHDL-LS seems to think the generate statements ends at the END of the BEGIN ... END block.

Take the following example:
```
  g_test: FOR i IN 0 TO 3 GENERATE
  BEGIN
  END;
  END GENERATE g_test;
```

Error reported on the END of END GENERATE:
  Expected ';'vhdl ls

Error reported on the GENERATE of END GENERATE:
  Expected 'library', 'use', 'context', 'entity', 'architecture', 'configuration' or 'package'vhdl ls

kraigher, 2022-12-07:
The above is not correct VHDL:

```vhdl
-- This is the correct syntax
gen: for idx in 0 to 1 generate
-- Declarative part
begin
-- Concurrent part
end generate;
```

Could you point to the part of the VHDL standard which says your code is correct?

kraigher, 2022-12-08:
Since you were so nice and helpful to place the standard definition before my eyes I did the work and fixed it for you.

closed 2022-12-08 by kraigher