VHDL-LS/rust_hdl #136: Create possibility to define the location of `vhdl_libraries`

delafthi, 2022-10-20:
Currently, `vhdl_libraries` can only be at two positions relative to the binary:

1. `../../vhdl_libraries`
2. `../vhdl_libraries` if feature `packaged` is set

It would be nice to provide the path as an argument to the `cargo install` command. However, I'm new to rust and don't know of a solution to achieve this behaviour. As such, it would be possible to use a buildscript to adjust the path and move the libraries to the defined location. Consequently, peoples would be less confused about the missing installed libraries (#113).

Currently, the sources have to be patched to change the path to wherever someone wants to install the vhdl libraries.

kraigher, 2022-12-07:
There is the VHDL_LS_CONFIG environment variable you can use:  https://github.com/VHDL-LS/rust_hdl/blob/master/vhdl_lang/src/config.rs#L287

delafthi, 2022-12-07:
I'm only talking about the installed libraries (std and ieee). Yes, `VHDL_LS_CONFIG` would be a possible solution. However, currently at https://github.com/VHDL-LS/rust_hdl/blob/master/vhdl_lang/src/config.rs#L236 the ls panics when it doesn't find the installed libraries. Conseqently, we would need to simply return when we add the installed libraries with `VHDL_LS_CONFIG`. Additionally, users would always need to define the `VHDL_LS_CONFIG` env variable for installed libraries, which isn't the best solution in my opinion.

Maybe the current patch I use for the AUR package describes better what I intend to achieve:

```patch
diff --git a/vhdl_lang/Cargo.toml b/vhdl_lang/Cargo.toml
index 166724f..4dbbff8 100644
--- a/vhdl_lang/Cargo.toml
+++ b/vhdl_lang/Cargo.toml
@@ -33,3 +33,4 @@ assert_matches = "^1"
 [features]
 default = []
 packaged = []
+installed = []
diff --git a/vhdl_lang/src/config.rs b/vhdl_lang/src/config.rs
index be9076a..6b61ed3 100644
--- a/vhdl_lang/src/config.rs
+++ b/vhdl_lang/src/config.rs
@@ -215,7 +215,9 @@ impl Config {
         let config_relative_path = if cfg!(feature = "packaged") {
             // relative to packaged bin folder
             "../vhdl_libraries"
-        } else {
+        } else if cfg!(feature = "installed") {
+            "/usr/lib/rust_hdl/vhdl_libraries"
+        }else {
             // relative to target/debug or target/release
             "../../vhdl_libraries"
         };
diff --git a/vhdl_ls/Cargo.toml b/vhdl_ls/Cargo.toml
index feb906a..d98db65 100644
--- a/vhdl_ls/Cargo.toml
+++ b/vhdl_ls/Cargo.toml
@@ -30,4 +30,5 @@ pretty_assertions = "^0"
 
 [features]
 default = []
-packaged = ["vhdl_lang/packaged"]
\ No newline at end of file
+packaged = ["vhdl_lang/packaged"]
+installed = ["vhdl_lang/installed"]
```

kraigher, 2022-12-07:
I does not seem elegant to add a cargo feature for every kind of package flavour. Why can't you use a symlink to solve it in your package?

kraigher, 2022-12-07:
It would be nicer not to even have the `packaged` feature flag and just try a list of known locations and emit a warning if multiple of them match. Such a PR I would probably approve.

delafthi, 2022-12-07:
The `vhdl_lang` binary is installed to `/usr/bin/` so relative to the binary the libraries would have to be placed in either in `/usr/` or `/`. I don't want to put the libraries there.

I agree symlinking would be a solution.

I just thought that installing the binary and libraries into the system might be a possibly future cargo task. The installed library position doesn't have to be variable and can be fixed (e.g. `/usr/lib/vhdl_libraries`).

delafthi, 2022-12-08:
Why do you provide the vhdl_libraries at all, shouldn't they be provided by the compiler/synthesizer in use? We would just need to provide a `vhdl_ls.toml` to define their location in the system.

kraigher, 2022-12-08:
@delafthi it really makes sense to have the VHDL standard library included out of the box and makes for a good user experience. That goes for IEEE as well since it is a de-facto standard library. 

Consider that synthesis tool or compiler might not even have these files on disk but just have them compiled into their binaries.
You might want to read VHDL code without having a simulator or compiler at all. 
If you have both a synthesizer and a compiler which of them should you take the files from?

To conclude; my wish is to not have many different cargo features for any kind of installation structure. I would rather the package feature be removed and the code instead automatically tried a few places where the files could be installed and warn on ambiguity. If you make such a PR I would probably accept.

closed 2022-12-14 by kraigher