VHDL-LS/rust_hdl #141: Compilation error on correct time constant syntax

Andy-Darlington, 2022-11-21:
A compile error is raised if there is no space between 1
constant TIME_CONSTANT : time := 1ns; incorrectly results in a compilation error.  A space must be added between the 1 and the ns.  
This impacts all usage of time variables, constants etc.  

[image]

Xcodo, 2022-11-21:
It looks to me like the examples of unit use in the VHDL 2008 standard all have spaces between the literal and the unit, so I think the error is correct (although some tools will accept code without the space).

maehne, 2022-11-22:
In VHDL, the space between value and unit is required. So, `rust_hdl` correctly flags an error.

Andy-Darlington, 2022-11-22:
Would you mind sharing the extract from the VHDL 2008 standard that states the space is required?  I'm surprised as I regularly use several compilation tools, this is the only one to flag this as an error.

GlenNicholls, 2022-12-07:
@Andy-Darlington simulators that support e.g. `1ns` likely did so because an older version of IEEE1076 was vague on the matter (I haven't checked to confirm, though). All new versions of simulators I use throw an error when there isn't a space, although they may have relaxed options/directives to ignore that issue. I found quite a few things in the 2008 LRM to support this.

*5.2.4.1* defines this explicitly and you will notice the required space between the abstract literal and unit name (`[]` denotes optional and note that `time` is a physical type):

```
physical_literal ::= [ abstract_literal ] unit_name
```

Where `unit_name` is defined as:

>unit name: A name defined by a unit declaration (either the primary unit declaration or a secondary unit declaration) in a physical type declaration. (5.2.4.1)

All uses of physical types in the 2008 LRM have a space beteen the abstract literal and unit name.

For `to_string`, they explicitly state a SPACE is required:

>For a value of a physical type, when forming the string representation for a TO_STRING operation, the abstract literal is a decimal literal that is an integer literal, there is no exponent, and there is a single SPACE character between the abstract literal and the unit name. If the physical type is TIME, the unit name is the simple name of the resolution limit (see 5.2.4.2); otherwise, the unit name is the simple name of the primary unit of the physical type. When forming the string representation for the WRITE procedure for type TIME, the physical literal is as described in 16.4.

And here's this, not explicit though:

>A physical literal consisting solely of a unit name is equivalent to the integer 1 followed by the unit name

For T'value(X) where X is a string and T is a physical type, it states a space is also required:

>...If T is a physical type or subtype, the parameter shall be expressed using a string representation of any of the unit names of T, with or without a leading abstract literal. The parameter shall have whitespace between any abstract literal and the unit name.

Andy-Darlington, 2022-12-19:
Thanks for looking into this.  Sorry for the slow reply on this one, it took me a while to get access to the standard.
Very useful to know, we'll update our source code to include the space.

closed 2022-12-19 by Andy-Darlington