VHDL-LS/rust_hdl #144: vhdl_ls shut down with ö/Ö ä/Ä ü/Ü ß in comments

std-rubic, 2022-12-19:
The vhdl_ls seems to crash/shut down with my files.

 {"jsonrpc":"2.0","id":1,"method":"textDocument/hover","params":{"textDocument":{"uri":"file:///a0.vhd"},"position":{"character":87,"line":482}}}
thread '
main' panicked at 'called `Result::unwrap()` on an `Err` value: Custom { kind: InvalidData, error: FromUtf8Error { bytes: [123, 34, 106, 115, 111, 110, 114, 112, 99, 34,
...
 110, 92, 114, 92, 110, 34, 125, 125, 125], error: Utf8Error { valid_up_to: 3738, error_len: Some(1) } } }', vhdl_ls\src\stdio_server.rs:30:23
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace

I use characters like ö/Ö ä/Ä ü/Ü ß in comments (round about 200 times). 
when I replace the characters with oe ae ue ss there is no crash/shut down.

I m working with notepad++ and NppLspClient plugin

kraigher, 2022-12-19:
Special characters work very well in other editors so I suspect that the notepad++ client violates the LSP protocol somehow. For example by not sending an correctly encoded string. Could you try to open the same project in VSCode with VHDL LS plugin, we know that one works?

maehne, 2022-12-19:
One thing, which could also play a role, is the actual character encoding of the file. Windows applications often still use the Windows-1252 codepage instead of UTF-8. If such files get interpreted as UTF-8, the umlauts typically don't show up correctly. Unfortunately, the VHDL standard [imposes ISO-8859-1 encoding outside of comments](https://insights.sigasi.com/tech/vhdl-file-encoding/), which can also contribute to the issue.

std-rubic, 2022-12-20:
Thank you, I converted the files to UTF-8 now it works.

closed 2022-12-20 by std-rubic