VHDL-LS/rust_hdl #145: Predefined attribute 'element is not recognized.

Holby, 2023-01-15:
Similar to #123.

Example, these declarations should be valid:
``` vhdl
type example_record_type is record
  data  : std_logic_vector;
  valid : boolean;
end record;
type example_vect_type is array (natural range <>) of example_record_type;

subtype example_8bit_vect is example_vect_type(0 to 1)(data(7 downto 0));
subtype example_8bit_elem is example_8bit_vect'element;
```
vhdl_ls returns two error messages on the last line:
`Expected ';'` and
`Expected 'library', 'use', 'context', 'entity', 'architecture', 'configuration' or 'package'`

kraigher, 2023-01-15:
Thank you for reporting. It should be easy to fix. I am in the middle of some refactoring now but I will push a fix in the near future.

kraigher, 2023-01-24:
Should be fixed in lastest release I just made. Please confirm by closing the issue.

Holby, 2023-01-25:
I can confirm that this issue is solved in the latest release, thank you!
I really appreciate all the work you do on this project!

closed 2023-01-25 by Holby