VHDL-LS/rust_hdl #149: Line not being parsed (no error check, goto definition etc)

guilbaudl, 2023-01-24:
Hello,
I found this issue where a line is not parsed at all. Here is a screenshot:

[image]

We can see that this is not highlighted as an error. With a correct use and import, the goto-definition is also not working.

[image]
[image]

On the second picture, we can see it works when out of the line.

I can't really help regarding this issue. The first `attribute` line works well, I can go to the definition of `string` for instance.

kraigher, 2023-01-24:
It is not clear to me what the problem is so I cannot reproduce it. Could you provide a small snippet of code that I can paste into vscode to reproduce the problem?

rhenescu, 2023-01-25:
Here is a small code that reproduces the issue:
```vhdl
entity testbench is
end;

architecture sim of testbench is
   constant C_RAM_STYLE : string := "block";

   type TYPE_RAM is array (0 to 3) of bit_vector(7 downto 0);
   signal ram : TYPE_RAM;

   attribute ram_style : string;
   attribute ram_style of ram : signal is C_RAM_STYLE;

   constant x : string := C_RAM_STYLE;
begin
end;
```

Go to definition and hover info is available on line 13:
[image]

but not on line 11 (move the cursor to line 11 column 49 and then press F12 or hover with the mouse):
[image]

kraigher, 2023-01-25:
Yes VHDL LS does not scan that part of attribute specifications yet. Attributes are not that common so I have not prioritized it yet. I will check if it is easy to scan the given the current state of VHDL LS. 

For some background:
VHDL LS has been built one step at a time from zero to becoming a parser to become type-aware by ignoring things that it does not handle yet. This is based on the design principle that it is more useful for an IDE to ignore things it does not yet understand than to give false errors. An IDE full of false errors is just irritating but partial checks and goto-definition still provides value.

kraigher, 2023-01-25:
I fleshed out the support for attribute specifications now so hover, goto definition and typechecking etc works for them. I skipped checking the entity class and saved that work for the future. This is now part of release 0.34.0 that is now building in Gitlab Actions.

closed 2023-01-25 by kraigher