VHDL-LS/rust_hdl #150: Errors reported when using generic types and generic packages

rhenescu, 2023-01-25:
The following errors are reported for the snippet below:

[image]

```vhdl
package stream is
   generic (type T);
   type T_STREAM is record
      valid   : bit;
      payload : T;
   end record;
end;

entity testbench is
end;

architecture sim of testbench is
   subtype T_DATA is bit_vector(7 downto 0);

   type T_PACKET is record
      last : bit;
      data : T_DATA;
   end record;

   package data_stream is new work.stream generic map (T => T_DATA);
   package packet_stream is new work.stream generic map (T => T_PACKET);

   constant C_DATA_STREAM : data_stream.T_STREAM := (
      valid   => '0',
      payload => "00001111"
   );

   constant C_PACKET_STREAM : packet_stream.T_STREAM := (
      valid => '0',
      payload => (
         last  => '1',
         data  => "10101111"
      )
   );
begin
   test : process begin
      report to_string(C_DATA_STREAM.payload);
      report to_string(C_PACKET_STREAM.payload.data);
      wait;
   end process;
end;
```

The code works with GHDL 0.37 and Questa Sim 2022.4:

```
../testbench.vhd:37:7:@0ms:(report note): 00001111
../testbench.vhd:38:7:@0ms:(report note): 10101111
```

rhenescu, 2023-01-25:
Not sure if related, but the following snippet also triggers a similar error for the string literal `"10"` at line 13:

```vhdl
package generic_package is
   generic (type t);
   subtype st is t;
end;

entity testbench is
end;

architecture sim of testbench is
   package gp is new work.generic_package
      generic map (t => bit_vector(1 downto 0));

   constant c : gp.st := "10";
begin
   test : process begin
      report to_string(c);
      wait;
   end process;
end;
```

This one crashes GHDL (see https://github.com/ghdl/ghdl/issues/2326), but works with Questa Sim 2022.4. EDA Playground doesn't work at the moment, so I can't test it with other simulators.

kraigher, 2023-01-25:
VHDL LS does not instantiate generic packages properly yet so it tries to ignore things from generic packages. You have found a place that currently should be ignored but was not.

VHDL LS has been built on step at a time from zero to becoming a parser to become type-aware step by step by ignoring things that it does not handle yet to still be useful. This is based on the design principle that it is better for an IDE to ignore things it does not understand than to give false errors.

Properly instantiating generic packages is not far away and actually something I plan to do in the near future.

kraigher, 2023-01-29:
I pushed support for instantiating generic packages to master. It is part of release 0.35.0 that is building in Github Actions right now. Please close the issue when you confirm it is working for your code.

rhenescu, 2023-01-31:
That's so cool, thank you! Had no issues with it so far, so I'll close the issue.

closed 2023-01-31 by rhenescu