VHDL-LS/rust_hdl #151: recursive vhdl_ls.toml

bundi78, 2023-01-27:
Hi,

I would like to request the possibility to "recursively" load the library definition using several `vhdl_ls.toml`.

My projects are heavily modularized and it would simplify the project handling if I could refer other toml files for example located in a sub-repo.

In addition to the current state, each toml file may contain one or more refers to other file, let's say `includes` or `imports`. The vhdl files paths in each toml file are always relative to their location.

With that feature, I could simply import any kind of sub-module or library without repeating the toml content. Also, if the sub-module change, the super project would still work without change.

kraigher, 2023-01-28:
Sure it makes sense and is easy to do. It's just sad that there is not standard way to define a VHDL-project structure so we have to build it for each tool.

bundi78, 2023-01-28:
Yes, it would be nice to have a common way since usually several tools are used during HDL development. One first step may be to update [VUnit](https://github.com/VUnit/vunit) too to additional read this toml file(s)?
[Sigasi](https://www.sigasi.com/) could also benefit from this. I proposed such a feature in the past during the early vs code extension development but no change so far.

kraigher, 2023-01-28:
People who use VUnit tend to prefer to have Python as the master source of truth for project definition interface and just have the `vhdl_ls.toml` files be exported from that automatically. There exists also `tsfpga` and similar and that is something you could to in your use-case. I did not really intend for `vhdl_ls.toml` to be a master file but rather be exported automatically from the main project definition so adding import features to it is not really something I thin should be necessary but I can do it anyway as a compromise to the current state of reality.