VHDL-LS/rust_hdl #153: Generic named "DEFAULT" gives false positive error

AndrzejKowalski9917, 2023-02-03:
I work with a legacy project and on some entities one of the generics is named "DEFAULT". This gives the following error message:

> Expected 'signal', 'constant', 'variable', '{identifier}', 'file', 'type', 'function', 'procedure', 'impure', 'pure' or 'package'vhdl ls

A minimal example to reproduce the behavior is the following:

```VHDL
library ieee;
use ieee.std_logic_1164.all;

entity test is
  generic (
    g_this_is_fine  : std_logic := '0';
    default         : std_logic := '0'  -- this is bad
  );
  port (
    clk   : in std_logic;
    reset : in std_logic
    
  );
end entity;

architecture rtl of test is

begin

end architecture;
```

It does not matter if "default" is written in capital letters or not. As far as I know, there is no reserved keyword "default" in VHDL. Xilinx ISE also has no problem with it.
Could it be, that it is a keyword in Rust which leads to this error?

By the way: Kudos for creating and maintaining this project. I use it daily and it makes working with VHDL a charm!

kraigher, 2023-02-03:
Unfortunately it is also a keyword for non-PSL:
```
interface_package_generic_map_aspect ::= [§ 6.5.5]
 generic_map_aspect
 | generic map ( <> )
 | generic map ( default )
```

kraigher, 2023-02-03:
I could add a 2002-compatibility flag at the library granularity in the `vhdl_ls.toml` file, how does that sound?

AndrzejKowalski9917, 2023-02-03:
Oh, I didn't know this. 
If time and effort for this is not to much, it would be a nice solution. Since it is an old project which I only need for reference, I could also just rename the generic to a valid name.

kraigher, 2023-02-03:
If you need it someone else probably also does. There is a lot of legacy code in this domain. I will add it in the near future. Can you confirm that it would work for you to set the 2002 compatability the library granularity? Adding it at the file granularity is more work.

AndrzejKowalski9917, 2023-02-03:
Yes it would work on a library level for me, since all files use an older VHDL standard.

matschi95, 2023-11-14:
Hello  @kraigher. We currently have the same problem. Are you planning to solve this in the near future? Or is it more work than previously assumed? 

btw: For us, it would also work at library level.

kraigher, 2023-11-14:
It is probably not much work but I have not priortized it so far. 

This is the type of issue where I would like new contributors to step up as it is an easy first task. 

I might do it myself as well if I feel the inspiration. As this is just an unpaid hobby project for me I cannot make any promises.

Schottkyc137, 2024-04-09:
This has been implemented as part of #284
While 284 does not fully support the 1993 standard, it fixes the issue mentioned here and enables switching the standard on the project level.
I would encourage anyone wanting better support for any language standard other than 2008 to open more targeted issues, for example like #275 with a list of features.

closed 2024-04-09 by Schottkyc137