VHDL-LS/rust_hdl #154: Generic list in subprogram not supported.

Holby, 2023-02-04:
Borrowing an example from [Ashenden's VHDL-2008: Just the New Stuff](https://books.google.be/books?id=ETxLguPMEY0C&pg):

``` vhdl
procedure swap
  generic ( type T )
  parameter (a, b : inout T) is
  variable temp : T;
begin
  temp := a; a := b; b := temp;
end procedure swap;
```

I get two errors on line two at "type":
"Expected 'is' or ';'" and
"Expected 'library', 'use', 'context', 'entity', 'architecture', 'configuration' or 'package'"

kraigher, 2023-02-05:
I will add support for it in the near future.

kraigher, 2023-02-11:
It seems not even GHDL supports this fully.  GHDL will complain that `T` is not defined.

Holby, 2023-02-12:
Yeah I found a similar issue for GHDL: [1470](https://github.com/ghdl/ghdl/issues/1470).
I tested the example with Questa, and it compiles without any errors.

kraigher, 2023-02-12:
Yes it is in the standard, it is just I cannot do everything at the same time so I have to assign priority. If it is not even supported in GHDL it lowers my priority :-)

PR #206 (2023-10-29): Parse subprogram header and parameter keyword
Closes #191, closes #154 

Now parses and analyzes the examples correctly.
closed 2023-10-29 by kraigher