VHDL-LS/rust_hdl #155: [Enhancement] Find references and definitions of architectures

pidgeon777, 2023-02-13:
Let's consider this `ENTITY_TOP.vhd` file:

```vhdl
library ieee;
use ieee.std_logic_1164.all;

entity ENTITY_TOP is
  generic (
    GEN : integer := 0
  );
  port (
    INP : in std_logic
  );
end entity;

architecture arch of ENTITY_TOP is
  signal sig : std_logic := '0';

  component ENTITY_1
    generic (
      GEN : integer := 0
    );
    port (
      INP : in std_logic
    );
  end component;

  component ENTITY_2
    generic (
      GEN : integer := 0
    );
    port (
      INP : in std_logic
    );
  end component;

begin

  ENTITY_1_1 : entity work.ENTITY_1(arch1)
  generic map(
    GEN => GEN
  )
  port map(
    INP => INP
  );

  ENTITY_1_2 : entity work.ENTITY_1(arch2)
  generic map(
    GEN => GEN
  )
  port map(
    INP => INP
  );

  ENTITY_2_1 : entity work.ENTITY_2(arch1)
  generic map(
    GEN => GEN
  )
  port map(
    INP => INP
  );

  ENTITY_2_2 : entity work.ENTITY_2(arch2)
  generic map(
    GEN => GEN
  )
  port map(
    INP => INP
  );

  PROC_p: process(INP)
  -----------------------------
  variable var_v : integer := 0;
  -----------------------------
  begin
      -----------------------------
      var_v := 0;
      -----------------------------
      if (INP = '1') then
        sig <= '1';
      else
        sig <= '0';
      end if;
      -----------------------------
  end process;
  -----------------------------

  process
  -----------------------------
  variable var_v : integer := 0;
  -----------------------------
  begin
      -----------------------------
      var_v := 0;
      -----------------------------
      if (INP = '1') then
        sig <= '1';
      else
        sig <= '0';
      end if;
      -----------------------------
  end process;
  -----------------------------
end architecture;
```

And two `ENTITY_1.vhd` and `ENTITY_2.vhd` files, respectively:

```vhdl
library ieee;
use ieee.std_logic_1164.all;

entity ENTITY_1 is
  generic (
    GEN : integer := 0
  );
  port (
    INP : in std_logic
  );
end entity;

architecture arch1 of ENTITY_1 is
  
  signal sig : std_logic := '0';
  
begin
  PROC_p: process(INP)
  -----------------------------
  variable var_v : integer := 0;
  -----------------------------
  begin
      -----------------------------
      var_v := 0;
      -----------------------------
      if (INP = '1') then
        sig <= '1';
      else
        sig <= '0';
      end if;
      -----------------------------
  end process;
  -----------------------------
  
  process(INP)
  -----------------------------
  variable var_v : integer := 0;
  -----------------------------
  begin
      -----------------------------
      var_v := 0;
      -----------------------------
      if (INP = '1') then
        sig <= '1';
      else
        sig <= '0';
      end if;
      -----------------------------
  end process;
  -----------------------------
end architecture;

architecture arch2 of ENTITY_1 is
  
  signal sig : std_logic := '0';
  
begin
  PROC_p: process(INP)
  -----------------------------
  variable var_v : integer := 0;
  -----------------------------
  begin
      -----------------------------
      var_v := 0;
      -----------------------------
      if (INP = '1') then
        sig <= '1';
      else
        sig <= '0';
      end if;
      -----------------------------
  end process;
  -----------------------------
  
  process(INP)
  -----------------------------
  variable var_v : integer := 0;
  -----------------------------
  begin
      -----------------------------
      var_v := 0;
      -----------------------------
      if (INP = '1') then
        sig <= '1';
      else
        sig <= '0';
      end if;
      -----------------------------
  end process;
  -----------------------------
end architecture;
```

```vhdl
library ieee;
use ieee.std_logic_1164.all;

entity ENTITY_2 is
  generic (
    GEN : integer := 0
  );
  port (
    INP : in std_logic
  );
end entity;

architecture arch1 of ENTITY_2 is
  
  signal sig : std_logic := '0';
  
begin
  PROC_p: process(INP)
  -----------------------------
  variable var_v : integer := 0;
  -----------------------------
  begin
      -----------------------------
      var_v := 0;
      -----------------------------
      if (INP = '1') then
        sig <= '1';
      else
        sig <= '0';
      end if;
      -----------------------------
  end process;
  -----------------------------
  
  process
  -----------------------------
  variable var_v : integer := 0;
  -----------------------------
  begin
      -----------------------------
      var_v := 0;
      -----------------------------
      if (INP = '1') then
        sig <= '1';
      else
        sig <= '0';
      end if;
      -----------------------------
  end process;
  -----------------------------
end architecture;

architecture arch2 of ENTITY_2 is
  
  signal sig : std_logic := '0';
  
begin
  PROC_p: process(INP)
  -----------------------------
  variable var_v : integer := 0;
  -----------------------------
  begin
      -----------------------------
      var_v := 0;
      -----------------------------
      if (INP = '1') then
        sig <= '1';
      else
        sig <= '0';
      end if;
      -----------------------------
  end process;
  -----------------------------
  
  process
  -----------------------------
  variable var_v : integer := 0;
  -----------------------------
  begin
      -----------------------------
      var_v := 0;
      -----------------------------
      if (INP = '1') then
        sig <= '1';
      else
        sig <= '0';
      end if;
      -----------------------------
  end process;
  -----------------------------
end architecture;
```

As you can see, in the `ENTITY_TOP` entity, instantiation of sub-entities occurs by specifying the wished architecture name. For example, for `ENTITY_1_1` instance:

```vhd
ENTITY_1_1 : entity work.ENTITY_1(arch1)
generic map(
  GEN => GEN
)
port map(
  INP => INP
);
```

When placing the cursor at `arch1` and looking for the definition, the new cursor position should be placed to the first character at `<a>`, in the `arch1` architecture word defined in the `ENTITY_1` entity:

```vhd
architecture <a>rch1 of ENTITY_1 is
...

Also, placing the cursor at `arch1` and looking for references, should report all of the lines where `arch1` is used as an architecture name, placing the cursor at `<a>` when jumping to the reference:

* ENTITY_TOP.vhd -> ENTITY_1_1 : entity work.ENTITY_1(<a>rch1)

It is important to note that when using `component` instantiation, such as:

```vhd
component ENTITY_1 is
  generic (
    GEN : integer := 0
  );
  port (
    INP : in std_logic
  );
end component;
```

then, it is not possible to specify the architecture to be used in the instantiation.

kraigher, 2023-02-14:
Should be easy to fix, I do it soon.

kraigher, 2023-02-14:
Will be part of release 0.58.0 that is building now.

pidgeon777, 2023-02-16:
Works good 👍.

closed 2023-02-16 by kraigher