VHDL-LS/rust_hdl #156: [Feature] Add missing LSP capabilities

pidgeon777, 2023-02-13:
Some very useful LSP capabilities currently missing, are:

'textDocument/documentSymbol'
https://microsoft.github.io/language-server-protocol/specifications/specification-current/#textDocument_documentSymbol

'workspace/symbol'
https://microsoft.github.io/language-server-protocol/specifications/specification-current/#workspace_symbol

'callHierarchy/incomingCalls'
https://microsoft.github.io/language-server-protocol/specifications/specification-current/#callHierarchy_incomingCalls

'callHierarchy/outgoingCalls'
https://microsoft.github.io/language-server-protocol/specifications/specification-current/#callHierarchy_outgoingCalls

kraigher, 2023-02-14:
Yes I know. I actually plan to add more LSP features now that the typechecking and overload disambiguation is very complete.

'textDocument/documentSymbol' and 'workspace/symbol' are probably my next project. After that I am going to look at the renaming refactor and then probably `[textDocument/completion](https://microsoft.github.io/language-server-protocol/specifications/lsp/3.17/specification/#textDocument_completion)`

pidgeon777, 2023-02-14:
Great news, by implementing those missing LSP specifications, yours will be the most complete VHDL LSP server available.

If you give me permission, I could open a separate issue for each of those capabilities, just to keep track of possible ideas/implementations, given that those are very different features.

kraigher, 2023-02-14:
This issue could be split into two issues as the *Calls and *symbol stuff is two different tracks.

kraigher, 2023-02-17:
The `workspace/symbol` is implemented now and part of the 0.62.0 release that is building in GitHub Actions.

pidgeon777, 2023-02-18:
Yes, I'm testing it right now, so far it seems to work well 👍.

kraigher, 2023-02-19:
I added the `textDocument/symbol` now as part of `0.63.0` as well as some improvements to `workspace/symbol`. It works well enough but there is some improvement potential. For example unnamed processes are not part of the document symbol tree yet. 

Also the LSP-protocol and VSCode puts som UX-limitations on this feature, https://github.com/microsoft/language-server-protocol/issues/1186 and https://github.com/microsoft/vscode-languageserver-node/issues/1194.

pidgeon777, 2023-02-19:
> For example unnamed processes are not part of the document symbol tree yet.

pidgeon777, 2023-02-19:
> I added the `textDocument/symbol` now as part of `0.63.0` as well as some improvements to `workspace/symbol`. It works well enough but there is some improvement potential.

First of all, great work 👏. I'm testing the new features right now.

> For example unnamed processes are not part of the document symbol tree yet.

Hopefully, those will also be added. `ctags`, when generating tags for an unnamed process, create an entry like the following:

```
process anon09185023
```

Where the number I suppose is randomly generated. And, if the process has a label:

```
process <LABELNAME>
```

kraigher, 2023-02-19:
Yes I plan to add unnamed processes as well. The only reason I do not add them yet is that the LSP-protocol requires a position for the document symbols and I do not compute a position on the parser for the unnamed processes.  

I could just add the position to the AST object now but I plan to rework the parser to keep all tokens instead which will be useful for comment extraction and auto format in the future.