VHDL-LS/rust_hdl #158: [Bug] Format of `textDocument/documentSymbol` LSP response doesn't allegedly match the LSP specification

pidgeon777, 2023-02-21:
[Neovim](https://neovim.io/) is a text editor tool which makes use of the Language Server Protocol (LSP) to provide code completion, diagnostics, and other features. The LSP specification defines a response format for the `textDocument/documentSymbol` request, and the one generated by `rust_hdl` seems not be be 100% compliant to the specification.

For example, let consider the following code:

```vhdl
library ieee;
use ieee.std_logic_1164.all;

entity ENTITY_TOP is
  generic (
    GEN : integer := 0
  );
  port (
    INP : in std_logic
  );
end entity;

architecture arch of ENTITY_TOP is
  signal sig : std_logic := '0';

  component ENTITY_1
    generic (
      GEN : integer := 0
    );
    port (
      INP : in std_logic
    );
  end component;

  component ENTITY_2
    generic (
      GEN : integer := 0
    );
    port (
      INP : in std_logic
    );
  end component;

begin

  ENTITY_1_1 : entity work.ENTITY_1(arch1)
  generic map(
    GEN => GEN
  )
  port map(
    INP => INP
  );

  ENTITY_1_2 : entity work.ENTITY_1(arch2)
  generic map(
    GEN => GEN
  )
  port map(
    INP => INP
  );

  ENTITY_2_1 : entity work.ENTITY_2(arch1)
  generic map(
    GEN => GEN
  )
  port map(
    INP => INP
  );

  ENTITY_2_2 : entity work.ENTITY_2(arch2)
  generic map(
    GEN => GEN
  )
  port map(
    INP => INP
[... 1005 lines cut ...]
}
```

Those are the tree different Neovim GUIs when using the three LSP servers:

`rust_hdl`:

[image]

`VHDL-Tool`:

[image]

`vhdl-linter`:

[image]

As you can see the [breadcrumb](https://code.visualstudio.com/docs/editor/editingevolved), which is displayed right at the top of buffer, is not correctly generated with `rust_hdl`.

Also, it seems that symbol scope information is missing in `rust_hdl`. You can see how the symbols panel report a yellow highlight for a node (symbol) and all of its parents (scopes).
This means that when the cursor is placed somewhere inside the buffer, then the curren scope is detected up to the highest-level parent node.

It is important to observe that when using other LSP servers for different languages, for example LUA or CPP, the breadcrumb and symbol scope information are always correctly detected, but that's not the case with `rust_hdl`.

In conclusion, it seems that in its current implementation, `rust_hdl` produces an output in a different format than the other LSPs, not allowing to make use of the breadcrumb and symbol scope features.

kraigher, 2023-02-21:
The LSP protocol defines two ways to answer the `textDocument/symbol` request: https://microsoft.github.io/language-server-protocol/specifications/lsp/3.17/specification/#textDocument_documentSymbol

I am using the new way, maybe your client only supports the old way. There is a client capability for this which I do not look at, I can disable this feature for clients that do not support the new format.

pidgeon777, 2023-02-21:
If possible, it would be great to test the version of `rust_hdl` which implements the old way, to verify if our hypotheses are right.

pidgeon777, 2023-02-21:
I can try to find out wich LSP standard version is implemented in the Neovim client.

pidgeon777, 2023-02-21:
> I could choose to either disable this feature for a client which does not support the new format or choose to implement the old SymbolInformation[] format.

It would be great to dynamically switch to the old `SymbolInformation[]` format according to the client's capabilities because by doing so it would be possible to use your LSP tool with other IDEs as well which implement an older LSP protocol on their client side.

kraigher, 2023-02-21:
I pushed a commit to master now that uses the old `SymbolInformation` structure if the client does not support the hierarchical document symbols. Please try and give report back.

pidgeon777, 2023-02-21:
Extremely quick! Just tried, but unfortunately, the same behaviour as before.

It is very strange because I also tried with LUS, C (clangd) LSPs, and all of them don't present this issue (breadcrumb always available and also symbol scope highlight).

kraigher, 2023-02-21:
Ok when I read your initial text carefully I see what the true problem is. The `DocumentSymbol` has two ranges and I fill in the same value for both, because I do not save have the other location information yet from the parser.

```typescript
/**
	 * The range enclosing this symbol not including leading/trailing whitespace
	 * but everything else like comments. This information is typically used to
	 * determine if the clients cursor is inside the symbol to reveal in the
	 * symbol in the UI.
	 */
	range: Range

	/**
	 * The range that should be selected and revealed when this symbol is being
	 * picked, e.g. the name of a function. Must be contained by the `range`.
	 */
	selectionRange: Range
```

This is not really a bug but an expected limitation. I do not compute the full source range of symbols today. This is actually something I am working on right now so hopefully it will become available soon. I want to reword how I store the source ranges a bit before I add more of it into the AST-tree in the current way I do it.

pidgeon777, 2023-02-22:
You're right, that would explain why the current LSP implementation of `rust_hdl` acts differently from the other LSPs.

We know that your current implementation supports hierarchy, defining parent and children symbols.

Situation is as follows, for `selectionRange`:

```
Parent1 [ start = 0, end = 7 ]
  Child1 [ start = 10, end = 16 ]
  Child2 [ start = 19, end = 25 ]
Parent2 [ start = 26, end = 33 ]
  Child3 [ start = 36, end = 42 ]
```

Actually, `start` and `end` delimit the starting and ending positions of the symbol.

Maybe as a temporary workaround, knowing the start and end of each symbol at the different hierarchy levels, we could make use of that information for example to determine the end of the scope for a given symbol?

For example, the values for `range` would become:

```
Parent1 [ start = 0, end = Parent2.start ]
  Child1 [ start = 10, end = Parent1.Child2.start ]
  Child2 [ start = 19, end = Parent1.end ]
Parent2 [ start = 26, end = <EOB> ]
  Child3 [ start = 36, end = Parent2.end ]
```

where `<EOB>` is the position of the latest character in the buffer.

Maybe this could be a quick and easy temporary workaround to integrate?

kraigher, 2023-02-22:
> Maybe this could be a quick and easy temporary workaround to integrate?

I can wait until it is done properly, it is just a waste to implement it with a work-around. If some clients renders the selection range on the editor it will also look strange.

Schottkyc137, 2024-04-27:
Some (but not all) should now have been included as part of #262

PR #302 (2024-05-16): Add source code location to all named entities
Closes #158 

Now adds a source span to all named entity declarations.
Also adds unnamed statements and declarations (i.e., process statements) to the entity hierarchy (visible, for example, in the outline in VHDL).

Any feedback concerning the format is greatly appreciated.
For example, I'm unsure whether `if` statements and similar should be part of the outline or not.
closed 2024-05-16 by Schottkyc137