VHDL-LS/rust_hdl #163: [Enhancement] Add diagnostic on never instantiated entities

pidgeon777, 2023-02-21:
Actually, when opening a VHDL entity which is not part of a project library, `rust_hdl` reports it.

It would be nice to also add a diagnostic message to the entities (and also their architectures for completeness) not instantiated in any of the other entities of the project.

The diagnostic level could be set to **WARN**, or maybe better **INFO**.

The diagnostic could be enabled or not, and its severity level set, by the user thanks to https://github.com/VHDL-LS/rust_hdl/issues/162.

kraigher, 2023-02-21:
Detecting unused code such as variables, signals, functions etc that are not read/written is something I hope to implement. There needs to be some side information that says a function or entity may be unused. For example if you are working on an IP-core library some entities might be unused within your own project and must be marked as "public"

kraigher, 2023-02-21:
For signals and variables as well as local functions it is easier as those can never be accessed from the outside unlike entities and functions in package headers.