VHDL-LS/rust_hdl #164: [Feature] Instance placement autocompletion

pidgeon777, 2023-02-21:
I know that autocompletion is planned, my proposal is to add instance autocompletion.

For example, by typing the name of an entity that you want to instantiate, an autocomplete suggestion appears for that entity and selecting it would let you instantiate it as a snippet with placeholders (for example with VSCode format `$1`, `$0`). Pressing `Tab` \ `Shift+Tab` would let you place the cursor at the points where a keyword is needed. Example:

[image]

SethGower, 2023-03-31:
@pidgeon777 I don't know what editor you use (it kinda looks like (neo)vim. If so, you could checkout this extension as a temporary solution to this: https://www.vim.org/scripts/script.php?script_id=3335

It lets you copy entity declarations and "paste" them as components/instances. I find it very helpful in my day to day, and have been using it for years. It's not the _most_ polished, but it does the job (mostly. Has some bugs though).

std-rubic, 2024-01-19:
A great feature, it seems to work with the 
https://github.com/VHDL-LS/rust_hdl/pull/244 
Thanks a lot !!

pidgeon777, 2024-01-23:
I have two proposal:

1. 

As you're aware, when defining generics and ports for a given entity in VHDL, they are arranged in an arbitrary order. For instance, the sequence might be `generic1`, `generic2`, `port1`, `port2`, `port3`, etc.

I propose that when entities are instantiated, the instantiated generics and ports should follow the same order as they were defined in their respective entities. So, autocomplete should output something as follows:

```vhdl
ENTITY_NAME: entity work.ENTITY_NAME
    generic map(
        generic1 => generic1,
        generic2 => generic2
    )
    port map(
        port1 => port1,
        port2 => port2,
        port3 => port3
    );
```

In the current implementation, it seems that the order is not preserved.

2.

An entity may have multiple architectures. In this case, the autocomplete should provide a list of architectures to choose from, put inside parentheses. For example:

```vhdl
ENTITY_NAME: entity work.ENTITY_NAME(ARCHITECTURE_1|ARCHITECTURE_2|ARCHITECTURE_3)
    generic map(
        generic1 => generic1,
        generic2 => generic2
    )
    port map(
        port1 => port1,
        port2 => port2,
        port3 => port3
    );
```

If the instantiated entity has only one architecture, then the parentheses should not be included (which is the actual implementation):

```vhdl
ENTITY_NAME: entity work.ENTITY_NAME
    generic map(
        generic1 => generic1,
        generic2 => generic2
    )
    port map(
        port1 => port1,
        port2 => port2,
        port3 => port3
    );
```

PR #315 (2024-06-30): Enable component instantiations
Closes #164 

With the recent additions and improvements to completions, component instantiations are now supported
closed 2024-06-30 by Schottkyc137