VHDL-LS/rust_hdl #166: Why no Cargo.lock ?

doronbehar, 2023-03-22:
Hello!

This program seems good, and I would really like to use it. In order to do this, I wish to package this software for NixOS. This requires patching this project to include a `Cargo.lock`. I was wondering whether you have intentionally disabled tracking it?

git blame: https://github.com/VHDL-LS/rust_hdl/commit/550335451183b4ba69d00f1994ffbff222b6b5bb

doronbehar, 2023-03-22:
> Why does NixOS require a `Cargo.lock` file?

Because Nix is a fully functional Package manager that builds all package in an isolated environment with no network connectivity. Including a `Cargo.lock` will enable us to prefetch all cargo dependencies before starting out the build, without updating the `Cargo.lock` file (now included in Nix' packages repository) after every update to this package.

> Are you aware that using `cargo install` does not work well with the VHDL-LS language serves as it requires the standard vhdl libraries to be located in a folder next to the library? #113

I am aware, and Nix has it's ways to fix such non portability issues commonly found in the wild. See:

https://github.com/NixOS/nixpkgs/pull/222395/files#diff-5bbe2cb800cf1147bd7006ddb93083f8435521058d2e3cbbe474449dfadc3b82R27-R28

Ideally, this option would have been read from an environment variable in addition to the defaults hard coded in `vhdl_lang/src/config.rs`.

kraigher, 2023-03-22:
You can make a PR to add the Cargo.lock if you need it.

PR #167 (2023-07-02): Start tracking Cargo.lock
Fixes #166 - helps packagers to build this package.
commit d928c56f (2023-07-02): Start tracking Cargo.lock

Fixes #166 - helps packagers to build this package.
closed 2023-07-02 by kraigher