VHDL-LS/rust_hdl #169: [Bug] Ignoring tb-folder

jakobjung10, 2023-04-05:
In my vhdl_ls.toml, i have this file:
`"src\\grpRs232\\unitTbdRs232Tx\\tb\\tb_TbdRs232Tx-Bhv-ea.vhd",`
(I use relative paths in my toml-file, but this is usually not a problem)

However, in the VS-Code-Client, the Language-Server detects the following error:
`c:\CHD4\Lab03\src\grpRs232\unitTbdRs232Tx\tb_TbdRs232Tx-Bhv-ea.vhd does not exist`

Maybe , the Language-Server ignores the "tb"-folder.

Is this a bug or some kind of feature??

Remillard, 2023-05-03:
That might actually explain something I'm seeing here.  I've got a testbench package with a number of procedures there named `tb_util_pkg.vhd` and it's explicitly called out in in the TOML file.  However when doing `vhdl_lang` I'm getting things like:

```
error: No declaration of 'write_note'
    --> C:\Users\nor71443\projects\onx-avx-fpga\common\src\discrete_csm\discrete_csm_tb.vhd:171
     |
169  |  
170  |          -- Test time.  Status words.
171 -->         write_note(dbg_stream, string'("-------------------------------------------"), C_DEBUG);
     |          ~~~~~~~~~~
```

And `write_note` is defnined in `tb_util_pkg.vhd`.  The section of the TOML file reads:
```
[libraries]
tb.files = [
    'sim/avalon/avalon_bfm_pkg.vhd',
    'sim/utility/tb_util_pkg.vhd',
]
```

I would hope it doesn't have any baked in ignoring, but that would explain why it's not picking up procedures in a package named that way.

kraigher, 2023-05-06:
There is no builtin ignoring of any files. I suspect the toml string contains escape characters such as \t etc.

Remillard, 2023-05-06:
I am pretty sure the string does not have that as I went through some variations with `*.vhd` instead of explicitly calling out that file.  However I will check and see when I get back to work on Monday.

jakobjung10, 2023-05-07:
> There is no builtin ignoring of any files. I suspect the toml string contains escape characters such as \t etc.

Indeed, my TOML-string contains the escape-character "\n" in every line.
Why is this a problem?
I only get this errors for 2 specific files, all the other files in the TOML-string do not cause an error...

Remillard, 2023-05-08:
Actually, I simplified things a bit and just blew away everything that might cause other issues.  The only thing the testbench is doing is calling `write_note` and it's an error in `vhdl_lang`.

[test_project.zip](https://github.com/VHDL-LS/rust_hdl/files/11421287/test_project.zip)

It still throws errors about OSVVM library but since I think the package is probably the source of the errors (i.e. `vhdl_lang` is not reading it for some reason) I didn't alter that file. 

In any event, that really points out something strange going on.

Schottkyc137, 2024-03-02:
I believe that this issue is similar to #251 with two libraries being defined for test benches and common files which was not intended. Can someone confirm this?

closed 2024-09-02 by Schottkyc137