VHDL-LS/rust_hdl #176: Add support for Xilinx xci files

axman6, 2023-06-05:
Xilinx's IP is managed using .xci files, which store the relevant metadata from configuring the IP in Vivado (as far as I understand at least). One of the components in of the XML file is the `boundaryDescriptionJSON` which describes the ports under `ip_boundary.ports`. From this documentation could be generated which describes the interface for these components in editors. 

I know it's probably a big ask, and I'm not sure if I can provide examples of .xci files if you don't have access to them, so mostly leaving this here as a place for discussion/search engine result when someone goes looking for support from vhdl_ls and xilinx's tools.

AndrzejKowalski9917, 2023-06-05:
When working with Xilinx IPs, I simply keep the .xci and the <component_name>_stub.vhdl file in version control. The *stub.vhdl file can than be added to the vhdl_ls.toml file.

Xcodo, 2023-06-05:
When you use IP core in Vivado the `.xci` file describes the IP and how you're using it. In our projects that is the file we check in to source control. If you use component instantiation in the module where you use the IP core then VHDL-LS does not give you an error, but if you use entity instantiation then you need to have a definition for the module included in your `vhdl-ls.toml` file.

Vivado generates a "stub" file for your customised IP core by default to give you the definition to make VHDL-LS happy. Once you have generated the output files the IP core's directory will contain a `*_stub.vhdl` file for the core with its ports. Add this to your `toml` file and this will provide the definition needed for VHDL-LS to be happy when using entity instantiation.

axman6, 2023-06-06:
Thanks for the quick responses - it looks like in our project we don't have the vhdl stub files, and it's not something I can add at the moment, which I was suggestion the feature since the info is there. But I can live without vhdl_ls supporting it, it's a pretty niche feature request.

closed 2023-06-07 by axman6