VHDL-LS/rust_hdl #179: [Bug] Encryption envelopes flagging many errors despite being valid VHDL

JHertz5, 2023-06-28:
VHDL-2008 introduced IP encryption directives, i.e.
```
`protect begin_protected
`protect <keyword> [= <value>]
`protect end_protected
```
I don't have access to the VHDL-2008 LRM, but this can be read about [here](https://firsteda.com/news/how-to-secure-your-ip-with-encryption/) and on p77-97 of [Ashenden's "VHDL-2008 Just the New Stuff"](http://ebook.pldworld.com/_eBook/FPGA%EF%BC%8FHDL/-Eng-/VHDL-2008.%20Just%20the%20New%20Stuff%20(Peter%20Ashenden,%20Jim%20Lewis).pdf)

An example snippet that raises errors in my project is as follows:
```
`protect begin_protected
`protect version = 1
`protect encrypt_agent = "XILINX"
`protect encrypt_agent_info = "Xilinx Encryption Tool 2020.2"
`protect key_keyowner = "Cadence Design Systems.", key_keyname = "cds_rsa_key", key_method = "rsa"
`protect encoding = (enctype = "BASE64", line_length = 76, bytes = 64)
`protect key_block
XpL1TFj1Z4ZooJGB3dSP6Pc8XBohs9jsfkhCnRPv/E0eBWI+lHNIXEa4u+PJkwlVZvWcONLhadzL
udIJCZSJZw==
```

This is automatically generated by Vivado from an IP block. With the upgrade to v0.65.0, VHDL-LS now reports many errors from the encrypted IP code in my project, mainly complaining about the backtick as an "illegal token", and about the key_block section, e.g. "Invalid bit string literal", "Invalid integer character 'l'", "Illegal digit 'C' for base 10".

I could find a way to exclude all of these files from the VHDL-LS project, but this is valid VHDL-2008 code and it would be very helpful if VHDL-LS could handle it. I don't even need them to be checked for correctness necessarily. I would be very satisfied if VHDL-LS just did not flag errors on these directives, and knew to ignore the blocks of text in the envelopes (i.e. after a `key_block`, `data_block`, or `digest_block` directive).

kraigher, 2023-06-29:
It makes sense to ignore these protected blocks. Obviously we cannot decrypt them without the private key.

JHertz5, 2023-06-29:
Yes, I definitely wouldn't expect the tool to be able to check the correctness of encrypted code! Ignoring the blocks would be ideal.

closed 2023-09-09 by kraigher

matschi95, 2023-11-08:
Hello, there are still errors with generated files from Xilinx. I don't know whether they comply with the standard or not.

The errors appear after the first  `protect key_block`. This means that the first one is ignored correctly, but not afterwards. Maybe `multiple key_blocks` are not implemented yet.

[image]

```
`protect begin_protected
`protect version = 1
`protect encrypt_agent = "XILINX"
`protect encrypt_agent_info = "Xilinx Encryption Tool 2021.2"
`protect key_keyowner = "Synopsys", key_keyname = "SNPS-VCS-RSA-2", key_method = "rsa"
`protect encoding = (enctype = "BASE64", line_length = 76, bytes = 128)
`protect key_block
lYvhEjj3nb5oH8uSNLeXMIy7nJYVR9CgwYrS2YsK1wH0yG7GgJF3h7LWVAsRpUASOB7rHmuPVhb5
Ot5CFu1eFeE97Zpvi2xwlrFd2yOm/xOs4mKX3gkTIBIJmAKj42AUYk/LR9j6mOwXFIQmoZqYXHak
Pq2yC2ljr0hY1gwTFtI=

`protect key_keyowner = "Aldec", key_keyname = "ALDEC15_001", key_method = "rsa"
`protect encoding = (enctype = "BASE64", line_length = 76, bytes = 256)
`protect key_block
Z+2GQWpqmewONlHVkL658DHQ1gOkrvPjxsrm0NDcBmt2DgE1WctRC0/WtmZNRR2P9xNPEc1AnD3g
x2bmQ9ClncBm4tJJUerktYV7SZWaAFXLpL0mImalEctnoiL1emAUpqT2xWqYmc7/Up4fedi3U63/
6fZpFkfLPe1f/3mRlu+DKs00gVRP+t6V+01C1oWFsyvdyS5tDx/D7YWjpI8AZn7PAxGanwdNWWSB
/kAFPcC2bUzb0T91+nSe2x7K7ugumFrWpHW6iiuiY86OlLeqrAD5SZsqHhPT9GqJmSzj5PdAcMm2
1N7wj661ojPTxlfvw7ydkwisxeQEZRQ1H8LwwA==
```

Enclosed you will find a test file (.txt-extension is needed  for attaching):
[fifo_generator_v13_2_rfs.vhd.txt](https://github.com/VHDL-LS/rust_hdl/files/13295395/fifo_generator_v13_2_rfs.vhd.txt)

PR #188 (2023-09-09): Feat: Tokenize tool directives
Closes #179 

Tokenize tool directives of the form
```
`identifier { arbitrary text }
```

In the current implementation status, this PR only tokenizes the tool directives. Any further processing is omitted for now. In the future, standard tool directives like the `protect` tool directive could be used for analysis