VHDL-LS/rust_hdl #183: Does not appear to support genericized packages

Remillard, 2023-07-31:
I noticed a June release and thought I'd investigate whether some of the issues I noted in April had been sorted (especially since Emacs 29 has been officially released, and I'd like to hook up a language server for VHDL via `eglot`.)

The test project that I delivered earlier shows another error that I didn't think much about figuring that it was configuraiton error, but I think this might be something more significant.

Running the command:
`vhdl_lang --config .\vhdl_ls.toml > vhdl_lang.log`

And then analyzing the logs, I note that as it goes through an analyzes OSVVM, a verification package delivered with Aldec Riviera-PRO (and also a project on Github) it's having issues with genericized packages:

```
related: Previously defined here
   --> C:\Aldec\Riviera-PRO-2022.04-x64\vlib\osvvm\src\ScoreboardPkg_int.vhd:52
    |
50  |  
51  |  
52 --> package ScoreBoardPkg_int is new work.ScoreboardGenericPkg
    |          ~~~~~~~~~~~~~~~~~
53  |    generic map (
54  |      ExpectedType        => integer,

error: A primary unit has already been declared with name 'ScoreBoardPkg_int' in library 'osvvm'
```

VHDL does allow for packages with generic maps to create new versions of the package with the constants bound to values at the time of instantiation.  I do not believe this should flag an error.

This small project also still exhibits the error of not picking up a package defined in the TOML file even though it's been declared.  The TOML file:
```
[libraries]
tb.files = [
'sim/*.vhd'
]
common.files = [
'src/*.vhd'
]
```

There's only a single file in sim/ which is `tb_util_pkg.vhd` and then when it scans the testbench source file it says:
```
error: No primary unit 'tb_util_pkg' within library 'common'
  --> C:\Users\nor71443\projects\test_project\src\discrete_csm_tb.vhd:7
   |
5  |  use ieee.std_logic_textio.all;
6  |  
7 --> use work.tb_util_pkg.all;
   |           ~~~~~~~~~~~
8  |  
9  |  -----------------------------------------------------------
```
[test_project.zip](https://github.com/VHDL-LS/rust_hdl/files/12215266/test_project.zip)

Schottkyc137, 2023-08-11:
For your second issue: Is `tb.files` and `common.files` what you want? The way you have specified it assumes that `sim/tb_util_pkg.vhd` is compiled into a library named `tb` and `src/*.vhd` is compiled into a library named `common`. However, you write `work.tb_util_pkg.all` which assumes that the package is compiled into the library work. So in my opinion, this should either be `use tb.tb_util_pkg.all` or (which I think is more likely) in your `vhdl_ls.toml`
```
[libraries]
work.files = [
'sim/*.vhd',
'src/*.vhd',
]
```

Remillard, 2023-08-11:
Ahh okay.  While it said libraries, I was not aware this was literally the VHDL libraries specified and not just a label for vhdl_ls.  This might also explain why it's not picking up OSVVM's libraries -- not entirely sure I called them OSVVM and might be named after the simulator.  I will try again today when I get an opportunity to play with the language server.  Thank you.

Schottkyc137, 2023-08-13:
Not sure if this helps but if you look at the [example project](https://github.com/VHDL-LS/rust_hdl/blob/master/example_project/vhdl_ls.toml), some OSVVM sources are commented out. Unfortunately I don't know why that is the case but it might be because of the same issue that you are facing. Concretely, `ScoreboardPkg_int_c.vhd` is not enabled, but `ScoreboardPkg_int.vhd` is.

kraigher, 2023-08-13:
OSVVM has duplicate packages for simulatorn that do not support generic packages. So you are Esther dupposed to add the generic package or the non-generic version.

closed 2024-04-09 by Schottkyc137