VHDL-LS/rust_hdl #192: [Feature] Add option to mute the error and/or disable lint checking

nselvara, 2023-08-22:
I'd appreciate a feature that allows to mute a warning/error and/or disable lint checking/code analysis for particular lines. Maybe to mute particular lines, we could add for example some kind of comments in the code.

```vhdl
-- lint_off
variable some_var: integer;
-- lint_on
```

Schottkyc137, 2023-08-26:
At the moment I don't think that this will be implemented. `vhdl_ls` is not a linter. If something is flagged as wrong, this means that it doesn't comply with the standard. Do you have a concrete use-case in mind?

nselvara, 2023-08-26:
Sorry, I didn't mean in particular the linter but the static code analysis. As an example this one could serve: https://github.com/VHDL-LS/rust_hdl/issues/184
Or this one: https://github.com/VHDL-LS/rust_hdl/issues/191

Schottkyc137, 2023-08-26:
Ah, I see. Yes, this would make more sense. However, I think that implementing this feature ist as hard or even harder than fixing the mentioned feature as there is currently no comprehensive way to get the tokens (and by that token [pun intended] no comprehensive way to get comments and/or tool directives for a certain line). An easier approach could be to turn on/off analysis for a file or a for a project completely. Would that be a solution?

nselvara, 2023-08-27:
I see, yeah I mean for the mean time that would be a sufficient solution. It's more like for developers who work on those projects could be a hint that the line is actually correct but the plugin flags it as an error. I'm not sure if that's maybe easier but what instead writing a comment but rather selecting lines of code and exculding those from the analysis. I think if the file/project can be excluded there must be some kind of flags that are stored so I assume that some lines of numbers can be somehow also included? I don't know hard this solution might be to implement.

darsor, 2023-10-19:
This feature would be very useful for PSL code, which is not currently supported by VHDL-LS.

PR #372 (2025-03-22): Conditionally enable tokenizing
Closes #192
closed 2025-03-22 by Schottkyc137